Ultimaker/Cura #12425: Initial layer height bug

MorningGlory, 2022-06-01:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ultimaker 2+

### Reproduction steps

1. Load a model into cura
2. Modify the initial layer height to anything except default setting
3. Preview layers

### Actual results

Strange patterns show throughout model, especially top layers

[image]

### Expected results

Slicing should look like this: 

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have not included a project file since it happens on all files, no log file since there was no crash.

GregValiant, 2022-06-02:
A project file will still be good.  I often print with a different Initial First Layer Height and I've never seen that problem.

MorningGlory, 2022-06-02:
By the way, I did try a fresh install of Cura 5 and still had the same issue

GregValiant, 2022-06-02:
I think it's a combination of certain Layer Heights with certain Initial Layer Heights.  I'm going to guess that something is going on when the last slice is in a certain relationship with the part.  I did check the model and it's fine.
I can reproduce what is in your image with .12 initial and .06 layers.  This is with .12 initial and .07 layers.
[image]
and this is with .13 initial and .06 layers
[image]

Try slicing it with Initial Layer Height of .12 and the Layer Height at .066 and then again at .067.  I get the "bad" image at .066 and the "good" image at .067.  So it looks like a break-off point there.

MorningGlory, 2022-06-02:
Thanks for the workaround Greg, I'll try it out!

GregValiant, 2022-06-02:
I've tried to reproduce this with my own models (produced in AutoCad) and I can't.  Now I know why I'd never seen the problem.
@MorningGlory What CAD software was that model produced with?

usernametakenaaaa, 2022-06-02:
Did you verify that your model is perfectly flat on the bottom ?
When looking at the first picture that was the first that came to my mind, that you have an error in the model within the magnitude of one layer and the layer height triggers it.

GregValiant, 2022-06-02:
I checked it as best I could (without having the actual CAD file) and it's flat.
When the STL "export utility" generates the triangles for the STL that feather edge of the threads may be causing issues with the calculation of the triangles.  The OP says it happens with all models though so my "feather edge" theory may be based on a red herring.  I never print at a .06 layer height and that may come into it as well.

MorningGlory, 2022-06-02:
The model was created in Onshape, here's my export settings:

[image]

GregValiant, 2022-06-03:
I think the number one improvement you could make would be to add a 0.5 chamfer or fillet to the top edge of the threads.  That would really help the engagement with the mating threads.  It would also get rid of the "feather edge" that I think is the problem at your current resolution.   I managed to put a .5 fillet on my model and at .05 layer height there weren't any problems.
Number 2 would be to go to a .07 layer height which seems to work OK.
If you drop the resolution setting one notch coarser than "fine" then the triangles in the STL will be coarser.  You have those threads at something like a .7 pitch.  If the triangles are coarser then you could loose definition on the threads.  I think the finest threads I've printed have been 1.0 pitch and with a .1 layer height they were fine.  I would leave lowering the resolution as a last resort.
I'm also curious to hear what the Cura team has to say about this since the way Cura is seeing the model appears to be involved.

nallath, 2022-06-10:
This very much looks like an issue that has to do with the slicing tolerance. You might want to check out that setting.

Joeydelarago, 2022-06-28:
> Did you verify that your model is perfectly flat on the bottom ? When looking at the first picture that was the first that came to my mind, that you have an error in the model within the magnitude of one layer and the layer height triggers it.

We have investigated this, and verified that it is happening on perfectly flat surfaces. 

This seems to be caused by the points on the top surface being directly on the z height of the cross section. There seems two be some inconsistent rounding up/down for these points. 

The team will discuss resolving this issue. 

Thanks! 🙂

Ghostkeeper, 2022-07-18:
Developers, see issue CURA-9462.