Ultimaker/Cura #12430: Slicing failed unexpected error

PhilBaz, 2022-06-02:
### Application Version

5.0

### Platform

win 10

### Printer

Voron 2.4

### Reproduction steps

try to slice with tree supports

### Actual results

fail

### Expected results

succeed 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Front re.zip](https://github.com/Ultimaker/Cura/files/8827572/Front.re.zip)

GregValiant, 2022-06-03:
Two things and they both relate to your base printer definition file.
You have the travel speed in cura at 450.  From the Voron base definition file:
"speed_travel":  { "maximum_value": 300, "value": 300, "maximum_value_warning": 501 }
So your travel speed is illegal per the printer definition that Voron wrote for your machine.

The definition file has this setting:
"meshfix_maximum_resolution":                   { "default_value": 0.01 }
I'm not sure what is illegal about that (it falls within the .001 and 3.00 limits) but Cura doesn't like it.  For reference my Ender 3 Pro needs that setting at about 0.4.
Change your Travel Speed to 300 and in Mesh Fixes change the Maximum Resolution to 0.10.  It sliced for me with those changes.

PhilBaz, 2022-06-03:
@GregValiant 

Hi Greg,

Ive altered the base definition file .json. It does not prevent me for slicing. I think its standard practice for Voron users as it can easily travel in excess of 500mm.

I think maximum_resolution is the default value. Can try changing it.

PhilBaz, 2022-06-03:
@GregValiant Yes, changing maximum_resolution to 0.1 di alow it to slice.....

And even just doubling it to .02 allowed it to slice as well.

Not sure if I should close this or not....is it a proper bug?

GregValiant, 2022-06-03:
Whether it's a bug or not isn't my call.  I just chase stuff around and see if something pops up.
In your case the travel speed thing stood out.  I changed the Max Resolution on a hunch.
This is from fdmprinter.def.json regarding the meshfix_maximum_resolution:
"default_value": 0.5,
                    "minimum_value": "0.001",
                    "minimum_value_warning": "0.01",
                    "maximum_value_warning": "3",
So on the surface your ".01" should have worked.  Someone from the Cura team will look at it.

nallath, 2022-06-10:
I've just tried to reproduce this on master and it sliced. We have changed the resolution algorithm, so it seems that the previous one had an issue that is now fixed.

closed 2022-06-10 by nallath