Ultimaker/Cura #12431: slice failure

Warped0ne, 2022-06-02:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

BIQU B1

### Reproduction steps

drag and drop STL into Cura 5
set to Standard (.2) profile
click slice

### Actual results

I get an error

Slicing Failed
Slicing failed with an unexpected error.  Please consider reporting the bug...

### Expected results

it should generate gcode that I can then save to a SD card

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

nothing additional

GregValiant, 2022-06-04:
That model has problems.
-> Reading file and indexing vertices
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
**--> 2 Non-manifold edges (?)
--> 3 Inverted faces (?)
--> 40 Degenerate faces (?)
--> 4 Duplicate faces (?)**
--> 0 Disjoint shells (?)
This is with the un-repaired model.  I re-selected the "Low Quality" profile and discarded all the changes.  There was something in there it didn't like.
[image]

Warped0ne, 2022-06-04:
I installed 4.13.1 and it has sliced 5 times now without an error.  I have attached the saved project for this STL in the same quality.

[image]<https://user-images.githubusercontent.com/64202104/171982122-63ddeefd-7a7f-4916-a22d-698eff888f41.png>

closed 2022-06-10 by nallath