Ultimaker/Cura #12432: Cura 5.0 Positioning is excruciatingly slow

Anathema-Device, 2022-06-03:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Custom Geeetech A10M

### Reproduction steps

Position two or more parts on the build plate.
Attempt to move one, it does not appear to move then the wait circle appears for about a minute (no kidding) then the move happens.
Similarly if you try to use the position dialog

Rotation does not suffer with this issue, just the xyx positioning
When unresponsive the CPU utilization goes to 15% but GPU utilization is 0

### Actual results

Big delay after trying to move or trying to type in the positioning fields

### Expected results

It is expected that positioning should be interactive, not with a 30 to 60 second delay for every change

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Let me know what docs may be helpful to diagnose the cause.

nallath, 2022-06-03:
Well, for starters logs would be nice. There should be some rudimentary performance data there that might indicate what is going on.

nallath, 2022-06-03:
Those logs are from a few days ago, and I only see a single instance in which an object was dragged, which was completed in 1ms (which is well within the range I would expect that to happen!)

The logs also indicate that you're using the 5.0.0 beta (and not the final release). The beta indeed had a performance issue.

So could you double check that you have the 5.0.0 final release and that the issue is still present there?

Anathema-Device, 2022-06-03:
Hi @nallath, I thought I had responded to this  and was about to add more but I see that my previous update is missing, maybe I was dreaming? hah

So, what I wrote was that I  had uninstalled the beta and installed the 5.0 GA version and the symptoms were different.  With the GA a hang was associated by a big spike in the GPU utilization and did not affect the zyx position input fields.  The current issue seems to raise the CPU utilization and affect dragging into position as well as coordinate entry in the position fields.

I went back and updated a couple of plugins to the latest and renamed the stderr and stdout log files then restarted Cura 5.  Now I don't currently see the lag when trying to position so maybe it was caused by one of the plugins that I updated?  Also I don't see any new logs generated so maybe they were from the beta only - I admit that I did not check them, sorry about that.

So in summary it would appear that the issue is not apparent right now.  Thank you for your interest in this.

Anathema-Device, 2022-06-06:
Well, let me revisit this and add some further information.
This morning trying to set up some prototypes on our 2.85 printer I notice that the lag is most severe first thing.  After restarting Cura a couple of times it seems to be cured?

Now this afternoon I switched to the config for the 8 channel MMU and the hang is worse than ever.  Seems to be amplified by the number of extuders?  It has been hung spinning for the last 10 minutes.

The log files seem to show nothing (0 bytes in size).  Is there a way that logging can be enabled to track this down?

nallath, 2022-06-07:
> The log files seem to show nothing (0 bytes in size). Is there a way that logging can be enabled to track this down?

The logging should be enabled by default. 

You can also try to run it with the extensive logging plugin: https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging. This will also capture profiling data. Do note that activating this plugin (extensions -> Extensive support logging - > start) will make Cura a lot slower, as it will gather a lot more data.

Anathema-Device, 2022-06-10:
I've installed the support logging plugin so hopefully this shows something.  This was a case where the delay was showing up when trying to position thing
[stderr_20220608.log](https://github.com/Ultimaker/Cura/files/8879368/stderr_20220608.log)
s

Anathema-Device, 2022-06-10:
And today I loaded a model that my wife asked for and tried to move it and Cura 5 ended immediately :
[stderr_2022.06.10.log](https://github.com/Ultimaker/Cura/files/8879379/stderr_2022.06.10.log)
(

nallath, 2022-06-10:
The support logger should create a zip file. That's the one I need.

Anathema-Device, 2022-06-10:
I found cura.zip in AppData/Local/cura is it what you are looking for?  It is ~50mb though.  If it is what you are interested in how can I get it to you - this tool seems to max out at half that :(

nallath, 2022-06-10:
If you run the remote support logger, it should also show a message as to where it saved the zip file. It should be called something like "remote-support_10-06-25-11_53_04.zip"

Anathema-Device, 2022-06-10:
[remote-support_22-06-10-11_14_26.zip](https://github.com/Ultimaker/Cura/files/8880082/remote-support_22-06-10-11_14_26.zip)
Ah, got it, sorry I didn't realize that I had to do something.
This last slice I was super careful not to try and move anything so I could get it started.  Does this show anything or do I need to restart Cura and reproduce the issue then gather a log?

nallath, 2022-06-10:
You have to start the profiler first. Then move some objects around and then stop it again. While it's running you should see a message in cura with a progress bar that bounces back & forth. 

After that you need to create the file in the same way as you already did before.

Anathema-Device, 2022-06-10:
[remote-support_22-06-10-12_56_39.zip](https://github.com/Ultimaker/Cura/files/8880779/remote-support_22-06-10-12_56_39.zip)
Hopefully I got it right this time.  I started Cura 5, then started the CPU profiler, loaded a simple part and tried to move it forward by dragging it with the mouse then I tried to move it back by changing the values in the position dialogue, then stopped the profiler & saved the result

nallath, 2022-06-13:
Yup. All the data is there now! I will discuss it with the rest of the team.

nallath, 2022-06-21:
We've added this ticket to our backlog. For our reference: CURA-9416

Anathema-Device, 2022-06-21:
Thank you!  I should add that changing to our MMU config (8 extruders) Cura is virtually unusable on the big Windows 11 machine so I have dug out a very old i7 laptop where it is just slow.