Ultimaker/Cura #12435: Cannot save to local disk and cannot open files from cura

alexban011, 2022-06-03:
### Application Version

5.0.0

### Platform

Linux (arch 5.18.1) (KDE Plasma)

### Printer

Anycubic i3 Mega S

### Reproduction steps

1. Opened Cura
2. Tried to open files
3. failed, so I opened dragged and dropped the file from file explorer (which KDE Dolphin if it helps)
4. sliced the model 
5. Couldn't save on local, but saving directly on removable disk worked 

### Actual results

as mentioned in reproduction steps

### Expected results

To let me select files normally and to be able to save locally as I don't like the way it names files

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Using AppImage. I had a bug so I did the following to fix it:
'''
./Ultimaker-Cura-5.0.0-linux.AppImage --appimage-extract
cd squashfs-root/
mv libstdc++.so.6 libstdc++.so.6_appimage
./AppRun
'''

akshimassar, 2022-06-06:
This seems to be the duplicate of https://github.com/Ultimaker/Cura/issues/12266

alexban011, 2022-06-06:
> This seems to be the duplicate of #12266

it is, I don't know how I missed it. I will close this now