Ultimaker/Cura #12436: Missing Print Setting

Firedogs2x, 2022-06-03:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Modix Big-60

### Reproduction steps

Hello, I was watching a video about supports, and following along with my CURA program open. When the video talked about the setting for the “Support Z Distance”, specifically setting the “Support Bottom Distance” I was not able to find that setting under “Printer Settings”.

So, I opened Preferences, Settings, and Settings Visibility. When I scroll down to the Support section, I see that option is checked.

So, I wanted to contact you and let you know what I found.

Thank you.
Dirk
[image]
[image]
[image]

### Actual results

I am unable to see "Support Bottom Distance".

### Expected results

I should see "Support Bottom Distance".

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

There is no more information beyond what I stated above.

Asterchades, 2022-06-04:
Support Bottom Distance is only available if you're using Everywhere for Support Placement. It's a completely irrelevant setting otherwise.

Firedogs2x, 2022-06-04:
Yes, I see that now. I did not have support everywhere.

Sorry for the mistake.

Thank you for your help and support.

closed 2022-06-04 by Firedogs2x