Ultimaker/Cura #12438: File Save Dialog on Ubuntu 22.04

bieber, 2022-06-04:
### Application Version

5.0.0

### Platform

Ubuntu 22.04

### Printer

Voron

### Reproduction steps

Running the latest 5.0.0 AppImage on Ubuntu (with the LD_PRELOAD fix from issue #12359) and I've noticed that the save file dialog after slicing is behaving oddly.  To reproduce, just slice a model and hit the "Save to Disk" button

### Actual results

First, instead of the file selector dialog popping up and taking focus, I just get a system notification that it's ready. 

[image]
 
Clicking the notification takes me to the file dialog, which is now missing the prefilled suggested file name

[image]

### Expected results

The file select dialog should pop up and take focus on its own, and it should prepopulate with a suggested file name

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing pops up in the terminal when opening the save dialog, but here's my command line in case there's anything helpful there

```
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 /home/bieber/bin/Ultimaker-Cura-5.0.0-linux.AppImage
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltimaTNDJry/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaTNDJry/share/cura/resources/qml/Actions.qml"
```

bieber, 2022-06-04:
I've also just realized that if I overwrite an existing file, I have to confirm it twice--one confirmation from the system file select dialog, and then a second confirmation dialog from Cura

AonoZan, 2022-06-09:
I dont even get notification, pressing button does nothing. I'm on Xubuntu so I probably don't have that type of notification popup, I have notification in top-bar but that gives nothing. Previous version of Cura 4.12 had every filedialog working, you could open files and save them no issue.

nallath, 2022-06-10:
Duplicate of #12266

closed 2022-06-10 by nallath