Ultimaker/Cura #12441: Cura 5.0 Error on Opensuse

nicolalandro, 2022-06-04:
### Application Version

5.0.0

### Platform

Opensuse Tumbleweed

### Printer

Irrelevant

### Reproduction steps

1. download new appimage
2. `chmod +x  Ultimaker-Cura-5.0.0-linux.AppImage`
3. `./Ultimaker-Cura-5.0.0-linux.AppImage`

### Actual results

[image]

### Expected results

Cura start correctly

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I reach that error
```
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libGL error: MESA-LOADER: failed to open crocus: /tmp/.mount_UltimaWeq22z/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib64/libLLVM.so.14) (search paths /usr/lib64/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open crocus: /tmp/.mount_UltimaWeq22z/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib64/libLLVM.so.14) (search paths /usr/lib64/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_UltimaWeq22z/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib64/libLLVM.so.14) (search paths /usr/lib64/dri, suffix _dri)
libGL error: failed to load driver: swrast
```

I have installed Mesa-devel  but I reach the same error.

plusline-morland, 2022-06-05:
I think this could be a duplicate to https://github.com/Ultimaker/Cura/issues/11856#issue-comment-box

Try
`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage`

libstdc++.so.6 could be located somewhere else on your system.
Check with `whereis libstdc++.so.6`

nicolalandro, 2022-06-05:
Thank you with that it works on opensuse

```
LD_PRELOAD=/usr/lib64/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-linux.AppImage
```

closed 2022-06-05 by nicolalandro