Ultimaker/Cura #12444: Fresh install of 5.0 unable to open

nxvash, 2022-06-04:
### Application Version

5.0

### Platform

Win10

### Printer

Ender 5 Plus

### Reproduction steps

installed cura 5 and tried to run it

### Actual results

cura opens to a white screen and goes not responding

### Expected results

cura should open like every other version has

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

steps taken:
Rebooted
uninstalled chitubox
rebooted again
uninstalled cura 5
rebooted

reinstalled cura 5
rebooted
uninstalled cura 5
rebooted
reinstalled cura 5 in a different path
rebooted

attached log:  [cura.log](https://github.com/Ultimaker/Cura/files/8838479/cura.log)

I have 2 32" LG 1440 144hz monitors attached to a Nvidia RTX Super

usernametakenaaaa, 2022-06-05:
Do you have two monitors ? Cura 5 currently crashes quite reliably on those.
Unplugging or disabling any secondary screens solves the issue.

nxvash, 2022-06-05:
> Do you have two monitors ? Cura 5 currently crashes quite reliably on those. Unplugging or disabling any secondary screens solves the issue.

Yeah, 2x 32" LG gaming monitors.  If that is the case I guess I will give up on 5 and go back to 4.10 since it works no problem without having to modify my whole setup. That's disappointing

nallath, 2022-06-07:
Duplicate of #12197

closed 2022-06-07 by nallath