Ultimaker/Cura #12445: Cura doesn’t load after opening

Musical-Jar, 2022-06-05:
### Application Version

5.0.0

### Platform

Mac 10.15.7

### Printer

Ender 3 Pro

### Reproduction steps

1. I was slicing a semi complex stl file (Fractal Pyramid) which was fine.
2. I left my computer go to sleep/closed my laptop until the next day. 
3. The next day it was frozen. When I clicked on the Cura window it made the “Mac noise” when it won’t let you do anything. 
4. I closed the app and then I couldn’t get the app to open again. 

### Actual results

When I click Cura, the blue window opens showing that it is loading everything. Then it closes and nothing pops up. There are no windows or the tools on the top left of the screen. The only thing there is the Cura drop down to either quit the app or open system preferences.

### Expected results

Cura should have opened a window with a new project after being opened. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://user-images.githubusercontent.com/106893317/172030354-e8518cda-67b1-420b-a3ce-c5f6d04533a5.MOV

nallath, 2022-06-07:
Please add the log files as requested in the template. There is currently not enough information for us to help you.

jeffqchen, 2022-06-13:
I've been having the same issue for a while. 5.0.0 is now totally broken and always start to this unusable state. All I did was using it for the first day. And it would never start again the next day.

Here's my log file.
[cura.log](https://github.com/Ultimaker/Cura/files/8887570/cura.log)

jeffqchen, 2022-06-13:
> Please add the log files as requested in the template. There is currently not enough information for us to help you.

I've been having the same issue for a while. 5.0.0 is now totally broken and always start to this unusable state. All I did was using it for the first day. And it would never start properly again the next day.

Here's my log file.
[cura.log](https://github.com/Ultimaker/Cura/files/8893530/cura.log)

nallath, 2022-06-14:
@jeffqchen If it worked once, that gives us some indication what it might be. If you remove (and backup by renaming it to something else) your preferences file ($HOME/Library/Application Support/cura/<Cura version>/cura.cfg), does the issue still occur?

jeffqchen, 2022-06-15:
> @jeffqchen If it worked once, that gives us some indication what it might be. If you remove (and backup by renaming it to something else) your preferences file ($HOME/Library/Application Support/cura//cura.cfg), does the issue still occur?

After renaming the existing cura.cfg, Cura starts fresh and went through its initial setup and then went to the main interface. A new cura.cfg was generated and all settings are lost. But that's expected.

nallath, 2022-06-16:
> On the first start it seems all settings are lost. But after fiddling with the machine settings and reconnecting to my printer, it seems to have brought back all the profiles and settings.

The cura.cfg also stores the currently active printer. So if you delete the file, cura creates an empty preference file and thinks that you don't have an active machine. This triggers the first run wizard. If you just add a random machine, it will continue

jeffqchen, 2022-06-18:
> The cura.cfg also stores the currently active printer. So if you delete the file, cura creates an empty preference file and thinks that you don't have an active machine. This triggers the first run [...]

It never died again. I've been using it for a few days and it started fine every time with the new cura.cfg.

I remember before it broke, I updated some plugins. Then it worked for the rest of the day, and broke on the next. Hope this information is helpful.