Ultimaker/Cura #12446: cura 5 crash Macair m1

Craw88, 2022-06-05:
### Application Version

5.0

### Platform

MAC OS monterrey 12.4

### Printer

Ender 5

### Reproduction steps

opening  multiple file  Cura ( more than 8 festal file) when reach file number 9 Cura crashed . 

### Actual results

system crash of opening 9 stl file opening up to 8 is o.k

### Expected results

not to crash 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[176(2).stl.zip](https://github.com/Ultimaker/Cura/files/8839195/176.2.stl.zip)
[173(8).stl.zip](https://github.com/Ultimaker/Cura/files/8839197/173.8.stl.zip)
 opening part 178 x 8 no issue Cura still work add in part 176 Cura crash

GregValiant, 2022-06-05:
What happens when you use the "Multiply" command instead of opening the file a bunch of times?

This is Cura 5.0 in Windows:
[image]

nallath, 2022-06-07:
Could you also include the log files? I've tried reproducing this issue (I did do that on Ubuntu, so it could be that that is the issue), but I was unable to do so.

closed 2022-06-21 by github-actions[bot]