Ultimaker/Cura #12447: Add {profile_id} into replacement patterns

haplm, 2022-06-05:
### Is your feature request related to a problem?

I'm automatically selecting Pressure Advance values by calling PRINT_START custom macro like this:

```PRINT_START BED={material_bed_temperature_layer_0} HOTEND={material_print_temperature_layer_0} CHAMBER={build_volume_temperature} FILAMENT={material_id} PROFILE={profile_id}```

I need both FILAMENT and PROFILE to select the right value, but unfortunately I seems that I can't access name of profile as a replacement pattern. 

### Describe the solution you'd like

Would you please consider adding `profile_id` into replacement patterns?

### Describe alternatives you've considered

There isn't an alternative, AFAIK.

### Affected users and/or printers

Everyone is going to benefit

nallath, 2022-06-10:
Simple enough to add! What should it place there? The name of the profile? Or do you really want the ID?

nallath, 2022-06-10:
Done! 

In 5.1 you can use {quality_changes_name} and {"quality_name"}.

The quality_changes_name is the name of the profile you created (if any). The quality name shows the name of the built in profile.

commit 73974972 (2022-06-10): Add quality_name and quality_changes_name to replacement patterns

CURA-9375
Fixes #12447
closed 2022-06-10 by nallath