Ultimaker/Cura #12448: Extrusion of filamet while at a retracted location

theffect, 2022-06-05:
### Is your feature request related to a problem?

Using Cura 5.0.0

I am printing a high and a narrow object (a bolt). The high speed at which the slicer codes the ender3v2 does not let the PLA cool down for enough time which in turns deforms the object.

Using the option "Travel->Enable Retraction" I was able to print the object at a much good precision though additional artifacts are created. This extra artifacts caused by the extrusion of filament at the retracted state which hangs there on the nozzle and going back to the object making the small amounts of extruded material at every retraction stick to the object.
There is also the effect of some stringing when the nozzle is retracted which also contributes to some imperfections.

### Describe the solution you'd like

I would suggest to pull the filament by the extruder just before retracting. It might reduce stringing and the extra filament hanging from the nozzle.

### Describe alternatives you've considered

as supports or adhesion rafts contribute to precision, I thought of adding a tower artifact that prints along side the target object. That tower object will be used to print during the retraction state needed for the target object to cool off. This might work as I have seen that printing a nut along side the bolt make it so that I do not need retraction cool time and no excess filament or stringing effects were visible.

### Affected users and/or printers

creality ender 3v2

### Additional information & file uploads

[3d-print-stuff.zip](https://github.com/Ultimaker/Cura/files/8839486/3d-print-stuff.zip)

GregValiant, 2022-06-06:
Have you tried printing 3 or 4 at once?  The additional movement and travel time can give the plastic time to harden without resorting to the "Lift Head" setting.

GregValiant, 2022-06-06:
Ender 3 Pro.  "Minimum Speed" 5mm/sec.  Layer Height .12.  Single 5015 cooling fan.  PLA at 205.
[image]

theffect, 2022-06-07:
> Have you tried printing 3 or 4 at once? The additional movement and travel time can give the plastic time to harden without resorting to the "Lift Head" setting.

Yes, I did. It went much better, that is why I suggested a nozzle cleaning tower object to be printed along side the target object. Though still, it did not come out as smooth as your output in the picture you posted.

I want to figure out the reason for the extrusion that is happening during the retraction period.

nallath, 2022-06-07:
I guess you want an ooze shield? That is a feature that Cura already has.

theffect, 2022-06-07:
> I guess you want an ooze shield? That is a feature that Cura already has.

mmm ok, could you have meant the "Draft shield" in the "Experimental" drop?

GregValiant, 2022-06-07:
Single-extruder printer here @nallath.

GregValiant, 2022-06-08:
The draft shield will work.  Anything that adds some time to the layer will help.
The other thing is your fan.  You didn't mention if it is the stock unit and I'm not familiar with the v2.  The layer cooling 4010 blower on my E3Pro was terrible.  It barely moved any air and what it did blow was aimed wrong.  I didn't want a monstrosity so I designed my own system around a 5015 and it puts the air where it should go.  I'm sure you can find something that will work.  It can be an excellent upgrade.

theffect, 2022-06-10:
I have stock ender3v2, but upgraded to dual z-axis, pei bed and better bed springs.
It seems that I do have the low end blower, I will investigate this aspect of quicker cooling.