Ultimaker/Cura #12450: Unable to open files from the menu or ctrl-o option

SnoBotics981, 2022-06-05:
### Application Version

5.0.0

### Platform

ChromeOS

### Printer

Crealtiy Ender-3 Pro

### Reproduction steps

1. Downloaded Ultimaker-Cura-5.0.0-linux.AppImage
2. Moved the image to the /usr/bin directory of the linux partition
3. Used chmod to change the file to +x (executable)
4. Started the program
5. Set things up for the Ender 3 Pro and Octopi
6. Tried to open a file and the file menu did NOT open
7. Downloaded the same version to a PC and things worked just find

### Actual results

Unable to open the file menu.

### Expected results

I would be able to open files from the file menu.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing further to add at this time.

SnoBotics981, 2022-06-05:
Seems very similar to the error, "Error file explorer Linux ChromeOS #12382"

SnoBotics981, 2022-06-06:
I was able to drag and drop files from my Files tool into the main page of Cura.

nallath, 2022-06-07:
Duplicate of #12266

closed 2022-06-07 by nallath