Ultimaker/Cura #12451: keine deutsche Übersetzung, kann Fehler nicht melden.

AnnoJoe, 2022-06-05:
### Application Version

5

### Platform

10

### Printer

FLSun SR

### Reproduction steps

Keine deutsche Übersetzung kann Fehler nicht melden.
Konfigurationsfehler, zurücksetzen ändert nichts. Meldung bleibt. Fehler-Text kann nicht kopiert werden.

### Actual results

?

### Expected results

?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

?

AnnoJoe, 2022-06-05:
?

Tampa, 2022-06-05:
Das ganze nochmal im ganzem Satz und spezifisch was genau der Fehler ist und wie man ihn erzeugen kann. In den Einstellungen lässt sich Cura auf mehrere Sprachen setzen, wenns Probleme mit der deutschen Übersetzung gibt dann stell dort einfach wieder English ein.
Ok now in a complete sentence and specifically what the error is and how to reproduce it. You can change the language in the settings if a translation is broken you can switch back to English.

nallath, 2022-06-07:
Sorry, our German is barely enough to go on a holiday and not get completely lost. Could you please report issues in English?

closed 2022-06-21 by github-actions[bot]