Ultimaker/Cura #12453: Cura 5.0 Appimage Fails to launch due to unhandled exception

richardwolfe4991, 2022-06-05:
### Application Version

5.0.0

### Platform

Void Linux - Vega64

### Printer

Did not get that far

### Reproduction steps

1. Try to execute Cura 5.0.0 Appimage - fails due to openGL issue (libstdc++.so.6 included with AppImage does not work)
2. Attempt to work around the library issue by preloading the system's libstdc++.so.6 library.

### Actual results

Upon attempt to execute Cura 5.0.0 AppImage with LD_PRELOAD of the system library - generates the following error in the terminal: 

`PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
[12164] Failed to execute script 'cura_app' due to unhandled exception!
`

### Expected results

Cura Appimage should "just work"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No log file has been generated yet. The Appimage won't start.

Elektronikus, 2022-06-06:
A similar problem exists with Kubuntu.
The error reported is: Cyclic dependency detected between "file:///tmp/.mount_Ultimax06kA5/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_Ultimax06kA5/share/cura/resources/qml/Actions.qml"
I have tried it with a not directly attached Ultimax Printer and it doesn't work also.

nallath, 2022-06-10:
If it gets to the point where it can complain about the cyclic dependency, log files should have been generated. Could you share them?

richardwolfe4991, 2022-06-12:
For what it is worth, I am not seeing anything about a cyclic dependency. Literally the only error I am getting is `Failed to execute script 'cura_app' due to unhandled exception!`. I checked the default location where log files are supposed to be generated on linux `home/.local/share/cura` and there are NO files there for Cura 5.0, and loose log files are EMPTY.

What Elektronikus reported is likely an entirely separate issue...

nallath, 2022-06-13:
> What Elektronikus reported is likely an entirely separate issue...

Agreed.

But unfortunately, without logs there is very, very little we can do :( Our own developers are also unable to reproduce this issue, which means that pretty much the last thing we might have to resolve this issue is also not available.

nirnex, 2022-06-14:
I'm encountering the same issue in Garuda (Arch variant) from what I've experienced and read it seems to be somehow related to AMD graphics cards. In my case a RX 6700 XT. When I first started using 5 during beta it worked no problem, but there was a set of updates to OpenGL and AMD drivers which seem to have caused it to break. Unfortunately when I first encountered it I didn't have time available to dig into it. It never gets far enough to produce logs However the libstdc++.so.6 does seem to be at least part of the issue as extracting the appimage and renaming the inbuilt copy to force use of the local version appears to work at least partially. Some functions such as signing into an Ultimaker account don't work.

richardwolfe4991, 2022-07-09:
Recently downloaded the Cura 5.1.0 BETA appimage to see if that fixed this issue. It did not.

HOWEVER. Just for curiosities sake, I tried running the appimage with root user (using sudo) 

`sudo ./Ultimaker-Cura_5.1.0-beta-linux.AppImage`

And it seems to be working fine. Running it without sudo, gives me an error:

`Could not probe OpenGL. This program requires OpenGL 2.0 or higher`

Seems like there is a dependency that Cura can only access with root privileges?

HarlemSquirrel, 2022-07-25:
This allows me to launch on Pop!_OS 22.04

```
QT_QPA_PLATFORM=xcb LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ~/Applications/Ultimaker-Cura-5.1.0-linux.AppImage
```

jellespijker, 2022-07-25:
Duplicate of #11856 fixed by running the `modern` AppImage.

The `QT_QPA_PLATFORM` is related to a different issue #12266 please continue that specific conversation unser that ticket

PR #12736 (2022-07-12): Cura 9205 appimage not launching on ubuntu 2204
Building two AppImages, one based on Ubuntu 20.04 and one with on Ubuntu 22.04. This also solves the sign in button not working on some Linux systems. Keep in mind that this is more of an intermittent solution and that we should still try to solves this in a more generic/future proof-way

Fixes both CURA-9342 and CURA-9205

The results of this run can be found here: https://github.com/Ultimaker/Cura/actions/runs/2648505664

- Fixes #11856
- Fixes #12453
- Fixes #11900
- Fixes #12232
- Fixes #12291
closed 2022-07-25 by jellespijker