Ultimaker/Cura #12454: Rotating the camera with the right mouse button makes the workspace disappear

Marcus9812, 2022-06-05:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3 Pro, Prusa MK3s

### Reproduction steps

Open Cura 5.0
In prepare mode or preview mode try to rotate the camera around build plate with or without a stl with right click 

### Actual results

The build plate/workspace disappears or changes to an unusual and irreversible shape until I restart Cura

### Expected results

The workspace rotates normally like cura 4.13

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://user-images.githubusercontent.com/64667719/172072852-edaf390b-1b8f-46e0-a3fc-e01d3e6af193.mov

And these options or switch the printer doesn´t help to recover the build plate 
[image]

nallath, 2022-06-10:
I've not been able to reproduce this. Could you share what camera settings in preferences you have set / enabled? 

What screen setup do you have (high DPI/ multiple screens?)

Marcus9812, 2022-06-13:
Hi, here are my specs: 
700 DPI in my mouse
Single monitor with resolution of 1440p
[image]
The settings are the same that I used to use in CURA 4.13 and older versions