Ultimaker/Cura #12455: Allow bridge skin density > 100%

traviselam, 2022-06-05:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Ender 3

### Reproduction steps

Attempt to set the bridge skin density > 100%

### Actual results

Unable to slice

### Expected results

Slice and allow the bridge skin density > 100% like Bridge second and third skin density.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This is technically a duplicate issue of a previously closed issue: 
- https://github.com/Ultimaker/Cura/issues/10736

And this was supposedly resolved in this commit:
- https://github.com/Ultimaker/Cura/commit/346640478227cb4e868d85ff37c4f5835cc5cc2f

However, in the latest CURA 5.0.0 this is not an allowed option. Was this a merge issue?

nallath, 2022-06-10:
Whoops! That was a mistake on our side. Should be fixed for 5.1 now!

closed 2022-06-10 by nallath

traviselam, 2022-06-11:
Awesome! I was doing some substantial bridge testing and I was hoping to get lower skin/wall flow with an increased density so I could cool less plastic over many passes to get a nicer-looking bridge first layer. Thank you!

commit e57af755 (2022-06-10): Remove incorrect maximum_value for bridge_skin_density

Whoops. That should have already been removed.

Fixes #12455