Ultimaker/Cura #12460: Bottom layer expands to additional layers

soxhi8, 2022-06-06:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

Going from Cura 4.13.1 to Cura 5.0.0, I discovered a weird slicing error on one of my test files.
1. Slice test file at 0% infill.
2. See weird bottom layer expanding additional 3 layers.

Project file and STL are attached.

### Actual results

[image]

### Expected results

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_Fasenplaettchen v3.zip](https://github.com/Ultimaker/Cura/files/8845541/CE3PRO_Fasenplaettchen.v3.zip)
[Fasenplaettchen v3.zip](https://github.com/Ultimaker/Cura/files/8845544/Fasenplaettchen.v3.zip)

GregValiant, 2022-06-06:
When I sliced the file with the Dynamic Quality profile and with the Standard Quality profile the layer aberration showed up.
Slicing with the Super Quality and Low Quality profiles it did not.
After playing around trying to find what was causing it (I did not figure it out) the aberration disappeared from the Dynamic and Standard quality slices.
But then it came back.

MariMakes, 2022-08-23:
Hey @soxhi8 

I had a look and it seems to be related with the amount of walls you have. 
These are different for different profiles and printers by default.
It makes sense that you started seeing this error starting from 5.0 since we changed the ways that walls are build up even though it's incorrect behavior. 
[image]

Can you let us know if ithis behavior actually impacted your printed part quality? 
That helps us prioritize the issue. Thank you!

soxhi8, 2022-09-12:
Hi @MariMakes - thanks a lot for the reply and the hint about the wall count!
I will do some more experiments when I have the time (which might not be for a while). As it stands, I am still using Cura 4 for the most part, because the unreliable seam-placement and this very problem kept me thinking about possible errors on complex models.
I do optimize for accuracy, so I would highly appreciate a fix. This one is not critical for me, but it keeps me away from Cura 5 for now.