Ultimaker/Cura #12462: alignment between mouse and object is incorrect

appelpop, 2022-06-06:
### Application Version

4.13.1

### Platform

Mac OS 12.4

### Printer

Ultimaker s5

### Reproduction steps

1. import file from fusion 360
2. arrange all models
3. select model
4. try to move object with arrows

### Actual results

1. model deselect
2. find the place where the arrow lights up
3. move object
[IMG_0060.pdf](https://github.com/Ultimaker/Cura/files/8846390/IMG_0060.pdf)

in the photo is my mouse and that is the location where the green arrow lights up

### Expected results

that I can click on the green arrow to move the object

it's the same with support blocker then you have to guess where to place it because you can't place it in the model but somewhere outside

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8846445/cura.log)

MariMakes, 2022-08-23:
Hey @appelpop, 
Thanks for reporting. 🎉  I'll have to bring it up with the team.
We are still trying to iron out some kinks with the latest version of Mac OS.

MariMakes, 2022-08-25:
Hey @appelpop,

I brought your issue up with the team. 
It might be related to moving Cura to screens with different resolutions. 

The next time you encounter this, can you try resizing your window?
This will trigger a redraw and it probably resolves your problem. 
https://user-images.githubusercontent.com/40423138/186631312-9f5ebb58-7895-4d36-9253-6c06428cd985.mp4

Good luck!

closed 2022-09-09 by github-actions[bot]