Ultimaker/Cura #12463: Unable to resize the print settings window

tillig, 2022-06-06:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Creality Ender 3

### Reproduction steps

1. Expand the Print Settings window.
2. Change any print setting.
3. Select the bar at the bottom of the window to expand it so you can see more settings.
4. Try to drag it up and down.

[image]

### Actual results

The resize bar doesn't move - you can't drag it up and down to resize the settings.

### Expected results

The bar should resize the settings window.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I was using a plugin in Cura 4 that was mounting settings to the sidebar to work around this since it's been a problem for some time, [but that plugin hasn't been updated for v5 yet](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/33).

nallath, 2022-06-10:
Some people have reported an offset between the mouse position on screen and where Cura thinks it is. Do you have a similar issue? Could you try if there is a spot close by where it does work.

If you hover over it, the mouse should change into a "drag" pointer.

tillig, 2022-06-10:
When I hover over it, it does change to the double-ended-arrow "resize" or "drag" pointer, but holding down the left mouse button when it changes and dragging up and down doesn't yield results. The mouse cursor moves but the resize doesn't happen. I can try doing a drag somewhere _near_ it without actually clicking on the bar, but the cursor won't be the right one. Maybe that doesn't matter. I'll give that a shot shortly.

tillig, 2022-06-10:
Just tried a bunch of stuff, none of it worked.

- Verified that the cursor changes to the move/drag double-ended-arrow only when the mouse is over the bottom bar.
- Tried dragging up/down at both ends of the bar, in the center, and in a few areas along the bar (horizontally). No luck.
- Tried dragging starting a couple of pixels above/below the bar (though the cursor was not the correct one) in the center, edges, and at a few points in between. No luck.
- Tried dragging starting in the middle of the bottom bar where the "Recommended" button is (though the cursor was not the correct one). No luck.
- Tried double-clicking on the drag area to see if I'd get lucky and have it "maximize" (not sure if that's a feature, but that'd be nice). No luck.
- Closed and reopened the settings window, tried all those things again. No luck.

I'm not sure what else I can try. This is pretty much the same as I found in later versions of Cura 4 and the only thing that rescued me was that plugin to pin the whole thing to the side of the screen and not have to fiddle with resizing.

All the other drag operations in the UI work - I can move the viewport around, I can move models around on the platform, etc. It's just this one resize issue, but it's kinda big because it makes the program unusable.

tillig, 2022-06-10:
Oh, I also tried dragging the settings window around to different locations on the screen to see if that'd make a difference. I am able to drag the settings window around without issue, but it did not affect the ability to resize it.

tillig, 2022-06-10:
I'm not sure where I even heard of that sidebar GUI plugin the first time, but I feel like it was on a forum somewhere that I was complaining about this and @fieldOfView popped in and said to work around it with that plugin. It's been quite some time and I've never looked back, so I didn't pursue it.

Are there any sort of persisted settings that may have gotten corrupted over time that I could go in and remove? Something to do with window position and size? Note I don't want to wipe everything out because I have custom materials and profiles and things I really don't want to lose.

Is there any trickiness around high DPI displays? I do usually have things scaled to 200% but I did try resetting to 100% and that didn't fix it. (Similar issues in things like Inkscape with high DPI and resize/menus.) Some sort of setting I should try?

Could there be a "double-click the resize bar and have the window maximize" sort of behavior added? I really just want the thing as tall as it'll go. I don't resize it other than to just make it bigger and then leave it. Maybe integrate that sidebar plugin right into the UI as a view option? (Totally spitballing, don't even know if that's possible.)

Any sort of debug info I could capture? I'm not a UI dev, to be honest, but I've done my fair share of debugging and troubleshooting over the years and may be able to do something like [capture events with Spy++](https://docs.microsoft.com/en-us/visualstudio/debugger/introducing-spy-increment?view=vs-2022) or whatever if that'd help.

nallath, 2022-06-10:
> Are there any sort of persisted settings that may have gotten corrupted over time that I could go in and remove? 

Maaaybe the preferences file, but it's a long shot. You can find it via help -> show config folder. If you move the "cura.cfg" file somewhere else, it should try to re-create the preferences. All other stuff will survive that (but as aways, be on the safe side and make a backup of the entire config folder!)

> Is there any trickiness around high DPI displays? I do usually have things scaled to 200% but I did try resetting to 100% and that didn't fix it. (Similar issues in things like Inkscape with high DPI and resize/menus.)  

That's pretty much the last thing I could think of. I'm pretty much grasping at straws here. 

Alternatively, you could also go into the preferences config file and change  the "settings_list_height" setting to something that is the right size for you. 

The sidebar plugin is pending review from us, so it will be back for 5.0 soon.

fieldOfView, 2022-06-10:
The sidebar GUI does not leave behind settings that would affect the normal GUI in this way.

While the new version of Sidebar GUI is in review, a prerelease is available here: https://community.ultimaker.com/topic/40313-fieldofview-plugins-for-cura-50/

tillig, 2022-06-10:
Weird thing for the day: I fired up Cura to do this test, then decided to try it on my second monitor (I have a dock that I sometimes use). I _left Cura open_ and docked.

Suddenly I'm able to resize that settings bar. And I can resize it on my regular laptop monitor or on the second monitor, so it's not a primary/secondary display thing. I undocked and can still resize.

I'm not sure what changed, if perhaps some setting got updated/overwritten when the second monitor appeared or what. I don't know if I've ever tried leaving the app open while docking, I think I was always either already docked or already undocked while running it.

In any case, super weird, but it looks like I'm fixed in some unintentional way.

Just to complete the circle, I did install the beta sidebar GUI plugin and that does appear to work, so yay!

Thanks for the help and the patience, I do appreciate it.

rubienr, 2022-07-08:
Issue still exists in version: 5.1.0-beta:
[image]

rubienr, 2022-07-09:
@tillig Is it possible to re-open the issue since it still exists in 5.1.0-beta?

tillig, 2022-07-09:
I don't mind reopening it as long as folks realize I won't be shepherding it through to completion or necessarily providing any more info than I already have.

rubienr, 2022-07-09:
Thank you @tillig!

rubienr, 2022-07-22:
Can confirm that issue still exists with [5.1.0](https://github.com/Ultimaker/Cura/tree/5.1.0).

rubienr, 2022-11-19:
Can confirm that issue still exists with [ Ultimaker-Cura-5.2.1-linux-modern.AppImage]( https://github.com/Ultimaker/Cura/tree/5.2.1).

trx1, 2022-12-17:
I also ran into this issue right after installing 5.2.1. I was able to use the workaround provided by @rubienr

rubienr, 2023-04-21:
Can confirm that issue still exists with [UltiMaker-Cura-5.3.1-linux-modern.AppImage](https://github.com/Ultimaker/Cura/releases/tag/5.3.1).

PWrzesinski, 2023-12-03:
I've just encountered this bug in 5.6.0 on a Macbook Pro M1. Workaround by @rubienr worked, thanks so much!

alexrec-alt, 2024-02-06:
Yep, that issue happen very randomly and not often, even restarting Cura does not help. BUT. 

@tillig mentioned trying to resize it on another monitor and it worked. As I had no second monitor atm I instantly tried to click "Restore window" then "Maximize" again (the middle one button between "Minimize" and "Close" window in **Windows**). And guess what - I was able to resize "Print settings" again. :)

version: 5.6.0

njsokalski, 2025-05-13:
I am having this problem in 5.10.0 as well. When I used the workaround provided by @rubienr is seemed ok (at least for now, who knows whether it will come back). I am using a multi-monitor setup as well, and the problem only occurred on my external monitors.

chrisglensmith, 2026-08-29:
Still getting this in Cura 5.9.0

pavelfedarenchyk, 2026-11-11:
still getting in Cura 5.14.0