Ultimaker/Cura #12466: Retraction stops on 16th layer

JerryNew99, 2022-06-07:
### Application Version

5.0.0

### Platform

windows 10

### Printer

Ender 3v2

### Reproduction steps

Loaded the 'retraction test' from 'part from calibration'. 
Set retract distance to .8mm and retraction speed to 50mm/s. I'm using a direct drive

       
[image]
 

### Actual results

The first 15 layers print fine, the retraction is occurring, no stringing. 
On the 16th layer the retraction just isn't occurring and there is string. 

### Expected results

Retraction should be occurring for the whole print 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8854708/cura.log)

GregValiant, 2022-06-07:
That's a plugin written by 5axes.  The [GitHub page is HERE](https://github.com/5axes/Calibration-Shapes).

JerryNew99, 2022-06-08:
I should add, I tried with another similar model from an external source.

GregValiant, 2022-06-08:
Did you use the post-processing plugin "retract tower"?

nallath, 2022-06-10:
> I should add, I tried with another similar model from an external source.

Okay, which one? 

It's very likely that this is caused by the "minimum extrusion distance window" setting.

JerryNew99, 2022-06-10:
I think you're right, thanks.

closed 2022-06-13 by nallath