Ultimaker/Cura #12468: ELEGOO NEPTUNE 2 SUPPORT

PeteyPub, 2022-06-08:
### Is your feature request related to a problem?

I have to add my printers manualy... 
and preview for ELEGOO neptune 2 CRASHES the printer on more complicated models. 
some preview normally... 

### Describe the solution you'd like

There was support for ELEGOO Neptune 2 and 2D in the OLD CURA 4.

### Describe alternatives you've considered

.

### Affected users and/or printers

ELEGOO NEPTUNE 2

fieldOfView, 2022-06-08:
> There was support for ELEGOO Neptune 2 and 2D in the OLD CURA 4.

Not in Ultimaker Cura. Perhaps you used a special version of Cura, or you added specific configuration files to your Cura 4 installation, but there was no support for the printers you mention in the official Ultimaker Cura 4 releases. Generally speaking, printer definitions that are included in the official Ultimaker Cura releases are never removed in subsequent versions.

nallath, 2022-06-08:
> Not in Ultimaker Cura. Perhaps you used a special version of Cura, or you added specific configuration files to your Cura 4 installation, but there was no support for the printers you mention in the official Ultimaker Cura 4 releases. 

Both of the definitions have been added 3 months ago: https://github.com/Ultimaker/Cura/blob/main/resources/definitions/elegoo_neptune_2.def.json && https://github.com/Ultimaker/Cura/blob/main/resources/definitions/elegoo_neptune_2D.def.json

But since the template hasn't been filled in at all, there is nothing we can do to help. Please fill in a bug template and provide us with all the info that it asks for (models that cause the issue, log files, specs of your computer, etc)

closed 2022-06-08 by nallath