Ultimaker/Cura #12474: CFBundleShortVersionString in the info.plist doesn't match the version of the App

paul-cossey, 2022-06-08:
### Application Version

5.0.0

### Platform

macOS

### Printer

N/A

### Reproduction steps

Browse to` /Volumes/Ultimaker-Cura-5.0.0-mac/Ultimaker-Cura.app/Contents/Info.plist ` the `CFBundleShortVersionString` shows as `0.0.0` I suspect this is probably just a typing error. 

### Actual results

This is the contents of the info.plist
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>CFBundleDisplayName</key>
	<string>Ultimaker-Cura</string>
	<key>CFBundleExecutable</key>
	<string>Ultimaker-Cura</string>
	<key>CFBundleIconFile</key>
	<string>cura.icns</string>
	<key>CFBundleIdentifier</key>
	<string>nl.ultimaker.cura.dmg</string>
	<key>CFBundleInfoDictionaryVersion</key>
	<string>6.0</string>
	<key>CFBundleName</key>
	<string>Ultimaker-Cura</string>
	<key>CFBundlePackageType</key>
	<string>APPL</string>
	<key>CFBundleShortVersionString</key>
	<string>0.0.0</string>
	<key>NSHighResolutionCapable</key>
	<true/>
</dict>
</plist>
```

Compare that to the info.plist of version 4.13

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>CFBundleDevelopmentRegion</key>
	<string>English</string>
	<key>CFBundleDisplayName</key>
	<string>Ultimaker Cura</string>
	<key>CFBundleDocumentTypes</key>
	<array>
		<dict>
			<key>CFBundleTypeExtensions</key>
			<array>
				<string>*</string>
			</array>
			<key>CFBundleTypeName</key>
			<string>Model Files</string>
			<key>CFBundleTypeRole</key>
			<string>Viewer</string>
		</dict>
	</array>
	<key>CFBundleExecutable</key>
	<string>cura</string>
	<key>CFBundleIconFile</key>
	<string>icon.icns</string>
	<key>CFBundleIdentifier</key>
	<string>nl.ultimaker.cura</string>
	<key>CFBundleInfoDictionaryVersion</key>
	<string>6.0</string>
	<key>CFBundleName</key>
	<string>Ultimaker Cura</string>
	<key>CFBundlePackageType</key>
	<string>APPL</string>
	<key>CFBundleShortVersionString</key>
	<string>4.13</string>
	<key>CFBundleSignature</key>
	<string>????</string>
	<key>CFBundleVersion</key>
	<string>4.13.1</string>
	<key>NSPrincipalClass</key>
	<string>NSApplication</string>
</dict>
</plist>
```

### Expected results

We (and many others) use the `CFBundleShortVersionString` to determine whether an app needs an update or not. As the latest version has a lower `CFBundleShortVersionString` the app will not update. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I've also noticed the signing identifier has changed to `identifier "nl.ultimaker.cura.dmg"` 

Thanks! 
Paul

paul-cossey, 2022-07-07:
Hi, 

I've just downloaded the 5.1 beta and noticed that the `CFBundleShortVersionString` is still showing as version `0.0.0` could it be bumped to something like `5.1.0`?

This is a deployment blocker for us, and likely anyone else that installs on macOS. 

I also noticed the signing has changed back to `identifier "nl.ultimaker.cura"` :-) 

Thanks
Paul

jellespijker, 2022-07-11:
Should be fixed for the 5.1.0 stable release with this commit https://github.com/Ultimaker/Cura/commit/c100f68f58d0c39b9f6b70df32edb2a4b3ca40c6

paul-cossey, 2022-07-11:
Brilliant, thank you! 

:-)

jellespijker, 2022-07-13:
Duplicaten of #12246 and fixed for 5.1 so I will close this ticket

closed 2022-07-13 by jellespijker