Ultimaker/Cura #12475: Cura 5.0.0 inconsistent font sizes

marcj71, 2022-06-08:
### Application Version

5.0.0

### Platform

Linux (Ubuntu 20.04.4 LTS)

### Printer

Ultimaker 3 Extended

### Reproduction steps

Cura 5 has some interesting choices for font sizes, at least as they appear on my screen. Cura 4 has much more consistency. Attached are images of the same windows for versions 4.13 and 5.0.0:
- Main window
- File menu opened
- Manage printers (preferences -> printers) window
- About window
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Actual results

Both 'on its own', and certainly when compared to Cura 4.13, Cura 5.0.0 has:
- A different font than the system font; I use Helvetica/Nimbus Sans
- A high menu bar (probably caused by increased whitespace everywhere)
- Tiny menu bar font -- or maybe it just looks small because other labes have become so big
- A large logo (top left) where branding takes up too much screen area (this is 'in your face', no subtlety at all)
- A remarkably high button bar (the one with the file button and material selection). The labels 'um3ext' and 'Normal - 0.15mm' are inconsistently large when compared to the material labels
- The menu items are spaced far apart and have tiny labels
- In the Manage printers window, the labels in the three buttons (Add New, Machine Settings, Connect..) are so large to be out of place. In 4.13, these were much more consistent with the other text.
- Also, the close button is gone in 5.0.0
- Surprisingy, the Close button in the About window looks like a link
- Windows with a scroll bar (such as Preferences) do not use the system scroll bar and the mouse wheel does not work in them

### Expected results

- Fonts should be more balanced, as they were in Cura 4
- System font should be respected, as it was in Cura 4
- All buttons should look like buttons, not a button here and a link there
- All windows should have a close button, all in the same corner. Having to use the window bar close button is unexpected, if only because it has the expectation of 'cancel' rather than 'apply and close'. How to I cancel any changes I may have made, other than loading the defaults?
- System scroll bars should be used, mouse wheel should work

I know there's a trend to make everything large with a lot of whitespace, but the overall feel is unbalanced.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no additional information

lorenzoromagnoli, 2022-06-08:
Hi @marcj71, thanks for this very detailed list, I'll look closely at all your suggestions. In the meantime,  could you try setting your screen scaling to 100% and let us know if the UI looks more balanced then?

I'm just asking because, while some of the thing you mentioned are due to totally new components (I.e. the menu bar which was using the the OS native application bar until 4.13 but we had to replace it with a custom made one in 5.0) some of the size differents you are reporting were for sure not intentional... and I actually cannot reproduce them 

This is for instance how I see them - left 4.13 and right 5.0
[image]

nallath, 2022-06-09:
> System font should be respected, as it was in Cura 4

This is actually intentional. Previously you would get a mix of fonts throughout the application. Now it always uses the same font everywhere.

marcj71, 2022-06-09:
@lorenzoromagnoli, I did not do any font scaling -- this is a standard Ubuntu Mate installation. These are my font settings:
[image]
Other than web pages, I see the same font in all applications except Cura.

lorenzoromagnoli, 2022-06-09:
Can you please check the "screen" scaling? 
Usually is a setting somewhere close to your display settings. 
[image]

marcj71, 2022-06-09:
New issue but probably related: Depending on the size of the window/monitor it is hard to select a material. The list just won't fit and it doesn't scroll. I wanted to select Ultimaker PLA Yellow, which is alphabetically challenged:
[image]

With a smaller application window, it gets worse:
[image]

Fortunately, I could reach it through the Settings menu:
[image]
but only because the menu starts higher. If this hadn't worked, I wouldn't have known how to select the material without Googling...

marcj71, 2022-06-09:
> Can you please check the "screen" scaling? Usually is a setting somewhere close to your display settings. [image]

I don't have any font scaling enabled. The Ubuntu Mate setting is 'automatic', which means it uses whatever is detected by the X server.
```
[    18.764] (II) modeset(0): Output eDP-1 connected
[    18.764] (II) modeset(0): Output DP-1 disconnected
[    18.764] (II) modeset(0): Output HDMI-1 disconnected
[    18.764] (II) modeset(0): Output DP-2 disconnected
[    18.764] (II) modeset(0): Output HDMI-2 connected
[    18.764] (II) modeset(0): Using spanning desktop for initial modes
[    18.764] (II) modeset(0): Output eDP-1 using initial mode 1920x1080 +0+0
[    18.764] (II) modeset(0): Output HDMI-2 using initial mode 1920x1200 +1920+0
[    18.764] (==) modeset(0): Using gamma correction (1.0, 1.0, 1.0)
[    18.764] (==) modeset(0): DPI set to (96, 96)
```
Is there a debugging option to see which fonts Cura selects?

lorenzoromagnoli, 2022-06-09:
As you correctly write in your last comment it seems that Ubuntu mate doesn't have a easy way to set a screen scale factor or a font scale factor. https://ubuntu-mate.community/t/scaling-in-mate/19091/14

But I still believe that most of the issues you are reporting are due to some wrong UI scaling. 
Let me try to answer to all your points one by one:

>- Cura Uses a different font than the system font; I use Helvetica/Nimbus Sans_
>- Fonts should be more balanced, as they were in Cura 4
>- System font should be respected, as it was in Cura 4

Cura now uses Noto sans practically everywhere. 
Also in Cura 4.13 and earlier, we used the same font, except for some dialogues and menus. Cura 5.0 brings the same font everywhere which is definitely an improvement when it comes to consistency. 
Most applications use custom fonts afaik (check i.e. vscode, slack, MS teams ). Shipping Cura with the same font on all OS allows us to have more control on how people are going to experience the SW. 

I agree with you that custom fonts should not be used in the main application menu which should actually follow the system theme styling, unfortunately, things broke during the qt upgrade and we couldn't find a way to have OS native application menu... so we replace it the native menu with a new menu with a Custom styling for all users on all OS.

>- A high menu bar (probably caused by increased whitespace everywhere)

Yes, it is indeed different from the menu bar of other applications for the reasons mentioned above

>- Tiny menu bar font -- or maybe it just looks small because other labes have become so big

To make Cura adapt to different people's screen scaling factors we measure the size of your default OS font and we derive a scale factory from that.
From your screenshots, it seems that your screen scale factor is set to something bigger that 100% in fact the stage and the menu look thicker than they should but the same scaling doesn't seem to apply consistently to the fonts. We need to look into that. 

>- A large logo (top left) where branding takes up too much screen area (this is 'in your face', no subtlety at all)

yes, it could be more subtle, but again if you are experiencing it as too big in 5.0 it is because of some weird UI scaling. 

>- A remarkably high button bar (the one with the file button and material selection). The labels 'um3ext' and 'Normal - 0.15mm' are inconsistently large when compared to the material labels

Same here, some weird scaling going on

>- The menu items are spaced far apart and have tiny labels

They could indeed be a bit more packed but it is mostly a matter of taste, I would rather go back into a scenario where the application menu follows the system styling than adjust the spacing.... we will never make everyone happy.  But yes, form your screenshot the font and the menu entry height don't look very proportionate... the same weird scaling.  

>- In the Manage printers window, the labels in the three buttons (Add New, Machine Settings, Connect..) are so large to be out of place. In 4.13, these were much more consistent with the other text.

Yes, we aligned all buttons to look similar and reuse the same components. The size of the button doesn't seem to be the issue here, but indeed the rest of the text in the UI is again not proportioned.

>- Also, the close button is gone in 5.0.0 
>- All windows should have a close button, all in the same corner. Having to use the window bar close button is unexpected, if only because it has the expectation of 'cancel' rather than 'apply and close'. How to I cancel any changes I may have made, other than loading the defaults?

Since we don't have a way to allow you to cancel your edits and go back to the previous state in any preference dialogue, we decided that it would feel more transparent to the user to just remove the close button which actually does exactly the same as closing the window. As I'm receiving multiple reports about this one, we might reconsider and bring the close button back. I'll discuss with the team  

>- Surprisingy, the Close button in the About window looks like a link
>- All buttons should look like buttons, not a button here and a link there

Fair enough... we replicated in Cura all the button styling that Digital factory uses, and there it was decided to have some very light tertiary buttons to compensate for the very heavy primary and secondary buttons. 
The thing is that tertiary should never stay on a page alone, but only be used together with secondary and primary. so yes, that close button shouldn't look like a link

>- Windows with a scroll bar (such as Preferences) do not use the system scroll bar and the mouse wheel does not work in them
>- System scroll bars should be used, mouse wheel should work

Indeed after the qt update, some scrollable areas started showing some issues. we have some tickets open about this, 

>- I know there's a trend to make everything large with a lot of whitespace, but the overall feel is unbalanced. 
Unfortunately, all the white space you are observing doesn't seem intentional, and doesn't show up on other os's, so we need to investigate a bit what is going on.

marcj71, 2022-06-10:
I only notice just now that the text in the Cura 5 About window ('End-to-end solution...') does pick up the system font in the right size! So, knowing it is possible... ;)

MariMakes, 2022-08-23:
Hey @marcj71 and @lorenzoromagnoli 
I'm going through our open issues and noticed this one is still open.

Do you still need me to bring it up to the team?
As you can see there are still a lot of issues open that our attention, I suspect that his one will not be on the top of our list any time soon.

closed 2022-09-06 by github-actions[bot]