Ultimaker/Cura #12477: Error Slicing the Attached Fille

jayh99, 2022-06-08:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender-3

### Reproduction steps

Imported the attached .STL, attempted to slice

### Actual results

Received an error Slicing failed with an unexpected error.

### Expected results

File would be sliced

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[alexa_r2d2_rasor_-_main_body.zip](https://github.com/Ultimaker/Cura/files/8863953/alexa_r2d2_rasor_-_main_body.zip)

GregValiant, 2022-06-08:
You have brought that model to the right place.  In Cura - after deleting the main body the other separate meshes sliced (that's pretty useless I know but it isolated the problem child)., So I exported the main body as it's own file.
Loading just that main body file - Prusaslicer, IdeaMaker, and Cura 4.7 sliced it.  I didn't know what to make of the IdeaMaker slice as it missed a LOT of detail.
Cura 5.0 and 4.13.1 would not slice it.
MS 3D Builder hung on it trying to merge all the separate meshes into one.
An online repair site indicated no errors in the main body and then hung up on it.

This is a view in MeshMixer.  The circled parts look odd (sort of a melted look and that back oblong hole is really weird) but as with Cura and the others; no errors were reported.  This is the main body only.
[image]

Here is a kind-of-the-same-view in Prusaslicer (also main body only).  The melted stuff is not present.  This is more how I expected it to look.
[image]

Here it is sliced with PrusaSlicer.
[image]

Sliced with Cura 4.7
[image]

GregValiant, 2022-06-08:
And here it is sliced in 5.0
[image]

I haven't been around here that long but in my short time it has always been an indicator of problems with the model when they will slice in some orientation but not others.  In this case I flipped it over and like magic it sliced.  I didn't try that the first time because "there are no errors in the model" but the repair programs hated it so I re-visited.  I would like to go on record at this point as being totally clueless about what is going on.  Good luck to all.

GregValiant, 2022-06-08:
I'm not having fun cuz it's driving me nuts.  Here it is sliced right side up.  The mesh modifier is a cutting mesh so nothing prints (or gets sliced) from 32.5mm off the build plate to 37.5mm off the build plate.  The zip is a project file.  I figure whatever it is that Cura doesn't like is within the zone that the mesh modifier is covering.
[GV_main_body complete.zip](https://github.com/Ultimaker/Cura/files/8865634/GV_main_body.complete.zip)

[image]

GregValiant, 2022-07-05:
I opened this in 5.1 Beta and it sliced with no problems.
I'll close this now as it appears that this bug has been fixed.
If you have further comments just post them here and I'll be notified.

closed 2022-07-05 by GregValiant