Ultimaker/Cura #12478: Won't open application on dual monitor setup.

KingBDS1, 2022-06-08:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ultimaker 5

### Reproduction steps

Whenever I open ultimaker on my dual screen setup, it looks like its open on an imaginary third screen but I can't get to it, move it, see it. It just shows its open on the taskbar.  So I have to unplug my 2nd monitor and restart the program in order for the application to come up. I have deleted and reinstalled the application twice. 

### Actual results

Opened once I unplugged second screen. 

### Expected results

Open no matter how many monitors I have. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Thats all I have. 
[output_log.txt](https://github.com/Ultimaker/Cura/files/8864755/output_log.txt)
[output_log.txt](https://github.com/Ultimaker/Cura/files/8864757/output_log.txt)

darrindossey, 2022-06-08:
It would help if Cura would just open any windows on the same monitor as the application is running on

nxvash, 2022-06-09:
good luck.  They will just close this and refer you to https://github.com/Ultimaker/Cura/issues/12197 which has 0 information on it other than basically unplug your second monitor.

nallath, 2022-06-09:
> good luck. They will just close this and refer you to #12197 which has 0 information on it other than basically unplug your second monitor.

That's because having duplicates just makes it hard to keep track. The issue is on our backlog so we will look at an actual fix at some point in the future (I hope that we will be able to look at it for the next release)

closed 2022-06-09 by nallath