Ultimaker/Cura #12479: Slice Failed

hhagquist, 2022-06-09:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

Opened the Sugar Skull from Hex3D
Went to slice using the Standard 0.2mm layer profile

### Actual results

Got a slice failed error
[image]

### Expected results

Should have sliced properly

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Could not upload project file. Find it on my Dropbox at https://www.dropbox.com/s/oi1iib4gcgt6wlf/Deluxe_SugarSkull%20R.zip?dl=0

GregValiant, 2022-06-09:
The model is watertight but there are errors.  After repairing it in MS 3D Builder it sliced.  The zip file won't upload because it's 47mb and is over the 25mb limit.
[image]

GregValiant, 2022-06-11:
@hhagquist I'm going to close this as it appears to be a model issue.  If you feel that it should remain open let me know.

closed 2022-06-11 by GregValiant