Ultimaker/Cura #12481: Tree supports hanging in mid-air

KamodaP, 2022-06-09:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Anycubic Vyper

### Reproduction steps

1. Import model
2. Set support to tree, 50% overhang, buildplate only

### Actual results

[image]
[image]
support hangs in mid-ai

### Expected results

Supports should be supported

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[bugshow_save.zip](https://github.com/Ultimaker/Cura/files/8868046/bugshow_save.zip)

DerGenaue, 2022-06-10:
Probably a duplicate of #8218.
If this is a critical issue to you, you can use [this Cura fork with improved Tree Supports](https://github.com/ThomasRahm/CuraEngine) in the meantime

Vandresc, 2022-10-14:
Though tree support still has some issues, this specific case has been fixed for the 5.2 stable release. 
[image]

closed 2022-10-14 by Vandresc