Ultimaker/Cura #12482: Vase mode / Z seam

Basti10785, 2022-06-09:
### Application Version

5.0.0

### Platform

Mac

### Printer

different

### Reproduction steps

Insert model
slice in vase mode

### Actual results

didn't work
[image]
k

### Expected results

print the model without Z sea
[image]
m

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Archiv.zip](https://github.com/Ultimaker/Cura/files/8868704/Archiv.zip)
3mf Files inside

GregValiant, 2022-06-09:
What I'm seeing here is that the 4.13 slice was incorrect and the inside diameter was ignored along with the "lock ring" feature at the top.  Those also comprise "outside" surfaces as there is air on one side and model on the other.  So the print you got was not the model you loaded.  It appears that "Remove all Holes" was invoked without it being turned on.  (That's a guess.)

The 5.0 slice accounts for the inside features and prints both the inner surface and the outer surface.  So "spiralize" didn't work because the nozzle had to move from the inside diameter to the outside diameter.  I think your line width at 1.2 was hiding what was going on and the inside diameter wall merged with the outside diameter wall.  You ended up with a single wall print but it was not spiralized and so there was a Z-seam.

To print this in spiralize in 5.0 turn on "Remove all Holes" in the Mesh Fixes section.  Then you will get the print that you got with 4.13 (and it will also be missing the inside features).

As an experiment, drop your nozzle size and line width to 0.4 in both Cura versions and slice in both versions again.  Turn off "Remove all Holes" in 5.0 and and check the preview, then slice it again with "Remove all Holes" turned on.

nallath, 2022-06-10:
Well, if you go through the layer view for a single layer, you do get pretty weird results. It first does the extrusion and then moves the nozzle back counter clockwise. 

This is also why the seam indicators are so obviously visible. It puts multiple stacked right next to eachother.

closed 2022-06-25 by github-actions[bot]