Ultimaker/Cura #12483: issues with bottom pattern in spiralize mode

lukk555, 2022-06-09:
### Application Version

5.0.0

### Platform

windows 10

### Printer

ender 3/ender 3 pro/ender 3 s

### Reproduction steps

1. load a model for spiralize mode,
2. select initial layer pattern "concentric",
3. select bottom layer pattern "lines". 

### Actual results

All bottom layers have pattern "concentric", providing poor bottom strength.

### Expected results

in Cura 4.13.1 same settings resulted in only the 1 bottom layer having concentric pattern, but all the next layers were printed in "lines" setting. This results with better bottom strength.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Not attaching any files - I was observing this issue with several different models.

GregValiant, 2022-06-10:
This has come up a couple of times since the release of 5.0.  It is the new default for spiralize to make all concentric bottoms.  Maybe @nallath could comment as I don't know why the developers made that decision.

nallath, 2022-06-10:
I'm not aware if this is a conscious decision on our end (it could be, but it's not something that I remember). But even if it is something that is done on purpose, I would argue that in that case the top/bottom pattern and bottom pattern initial layer should be hidden in that case, as they don't do anything. 

I will add it to our long list of issues to discuss. Thanks for reporting it!

lukk555, 2022-06-10:
Thanks guys for your quick reply! I'd really appreciate if you can go back to the previous config here, as the robustness of spiralized prints suffers significantly on this one. I often do prints in vase mode with 0.28mm layer height and 0.8mm trace width (0.4mm nozzle). Print like this comes quite solid in PETG when I do it with initial bottom layer as concentric and then 2 layers as lines. When I print all 3 bottom layers as concentric is much more fragile. Now I need to use 4.13.1 for spiralized prints.
On the upside: I like the movement of nozzle on concentric layer better on 5.0 than on 4.13.1, but would like to have a choice.
Cheers!

nallath, 2022-06-13:
We found the original place where this was reported. As such we will close this ticket in favor of that one

Duplicate of https://github.com/Ultimaker/CuraEngine/issues/1577