Ultimaker/Cura #12486: Cura won't open after a restart

nixz-leon, 2022-06-09:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

ender 5

### Reproduction steps

I had the same issue with the previous version of cura, I tried to uninstall, install to a different drive, installed the qt dev kit. 

### Actual results

It should work after I restart my computer but it flashes a white screen and exits the application

### Expected results

it should just open up the window 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8873495/cura.log)

nallath, 2022-06-10:
Do you have a secondary monitor? We've had multiple reports that this can cause Cura to not start.

Ergonomicmike, 2023-07-13:
I am having a similar problem, although different. I am running Cura 5.4.0. When I slice with Organic Supports and close the problem after that, it will not re-open fully. (Windows 10.) I see the splash screen and many things load. But then, at the last minute (presumably when Cura is reloading the model), the program closes.

As a workaround, I have reinstalled Cura (manually) by unzipping to a new folder. 

Please let me know if I should open a separate issue for this. And/or  what log files you need. (Assuming log files are created during a failed restart.)