Ultimaker/Cura #12487: "eject" button still corrupted disk

Throneaway420, 2022-06-10:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

Saved file to removable disk.
Clicked "eject" on Cura

### Actual results

Printer reads "SD init fail", drive is corrupted and needs reformatted. 

### Expected results

After clicking eject, drive should not become corrupted. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Thanks for destroying all the data on my drive.

GregValiant, 2022-06-10:
I have the same printer and the same operating system.  I have not seen or heard of this behavior.  I don't know the inner workings of Cura but in my own software I make calls to Windows when dealing with drives.  One of them happens to be an Eject function but I just ask Windows to eject the drive.  Did you try re-formatting the SD card?  I still use the card that came with my printer and I have had to reformat it several times as it has become corrupted through normal use.  It's a popular problem with the poor quality memory cards that Creality includes with their printers.

nallath, 2022-06-10:
Cura does exactly the same. We call the windows API to eject SD cards. 

As such, I don't see how this would be a Cura issue (instead it's probably a poor quality SD card)

closed 2022-06-10 by nallath