Ultimaker/Cura #12489: Slicing Failed

RoadyRunner, 2022-06-10:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

1 : Load the stl.
2 : Only Slicing,
3 : and then "Slicing Failed".

### Actual results

 "Slicing Failed"

[image]

### Expected results

Slicing ok

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Grand Rond jante Bug.zip](https://github.com/Ultimaker/Cura/files/8878752/Grand.Rond.jante.Bug.zip)

nallath, 2022-06-10:
I've been able to confirm that this still happens with the latest version of the engine. 

Interestingly enough it doesn't crash if you discard your current changes. I've not looked into which setting it causes it, but if you're feeling particularly helpfull, it would save us a lot of time if you could find out what setting causes it! 

I will bring the issue up with the rest of the team regardless.

Asterchades, 2022-06-11:
Take a look at the profiles in use for this, and the two issues I referenced from this as well. All three show thousands of values (including assigned values, limits, enabled, etc) with a leading equals symbol despite all three being different printer configurations. This in turn is resulting in "blank" values being set in the UI, which then get passed to the slicer that obviously then fails.

Once is an accident, twice a coincidence, but three times is an act of war - so to speak.

Could this perhaps be an issue with the profile importer or version upgrader used in 5.0? Or is 5.0 simply less tolerant of these weirdly constructed profiles?

Triangulix, 2022-06-11:
It wouldn't be a problem for me to create a new printer profile and manually carry over my most important print profiles, but... #9253 and #12419

Triangulix, 2022-06-11:
I searched through my long-time backups and traced these entries back to Cura 3.5. They have beeen carried over through all 3.x and 4.x versions.
My conclusion is that someone changed the syntax of these configuration files for version 5 and forgot to add a converter.

Asterchades, 2022-06-12:
First off, my apologies. The superfluous equals symbol appears to be a result of the project exporter, which the project importer seems to handle just fine. So that's clearly not the issue (though it is a curious behaviour on the part of the project system).

That being said... I have found two separate settings which can correct all three projects in question, and I suspect the issue may lay in how the two features interact with each other. If you adjust the Wall Line Count (either directly or via the Wall Thickness setting) to a different value then all three will slice correctly. I'm reluctant to say "any other value" but every value I've thrown at it so far works. The other setting is Infill, which if set to 0% (not practical but useful for diagnostics) will allow all three models to slice successfully. 

I've also identified a different setting on all three projects other than the above which allow them to slice successfully. The model "Grand Rond jante Bug" can be made to work by setting the Top Skin Expand Distance to at least 0.275mm, "Hooks" from #12495 will work if you set the Infill Line Multiplier to the default 1, and "Druckkopfhalter Cura" from #12495 will work if the Infill Overlap is set to 29.874% or lower.

However I'm becoming more reluctant to consider #12491 and #12495 as Cura issues; though it could perhaps handle them more gracefully. While all three of these projects contain broken models, if those two are corrected using 3D Builder then they'll slice perfectly without any settings being changed. "Grand Rond jante Bug" still requires one of the above settings to be altered even after addressing the model.

Triangulix, 2022-06-12:
> First off, my apologies. The superfluous equals symbol appears to be a result of the project exporter, which the project importer seems to handle just fine. So that's clearly not the issue (though it is a curious behaviour on the part of the project system).
No, it is Cura's internal syntax for calculated values. It is used everywhere in the configuration files and also in the end block of GCODE files where the settings are stored as comment.

rburema, 2022-08-19:
Tested the initial .zip the OP posted on an early alpha build of 5.2 -- slicing succeeds now. Closing this. If this (eventually, when it's out) still occurs in 5.2 beta or later, please open a new ticket (or ask us to reopen this one).

closed 2022-08-19 by rburema