Ultimaker/Cura #12491: Engine crashes when slicing this project

Triangulix, 2022-06-10:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Create a project like mine and try to slice it.

### Actual results

The slicer engine crashes.

### Expected results

A sliced print job.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Druckkopfhalter Cura.zip](https://github.com/Ultimaker/Cura/files/8881171/Druckkopfhalter.Cura.zip)

Asterchades, 2022-06-11:
Your models are in dire need of repairs. There's over 93 _thousand_ open edges. Despite that I'm still able to slice successfully using both my own custom profile and the default "Custom FFF printer" profile.

That said, there's problems in your printer configuration (custom.def.json). Since this is a custom configuration you're going to need to either source a repaired configuration from the original author or fix it yourself, as these will be setting invalid parameters. A blanket search/replace for "= to replace with " should get most, if not all of them, which in turn will allow this - and likely any other model - to slice correctly.

Example of the type of setting I'm talking about:
`#1417 "value": "=0 if infill_sparse_density == 100 else math.ceil(round(top_thickness / resolveOrValue('layer_height'), 4))", `

That leading = in the assigned value will result in the setting (in this case, the Top Thickness) being given a blank value. Some settings may incorrectly allow blank values (many, if not all of which are being allowed in this profile by the same superfluous equals symbol) which in turn crash the slicing engine.

closed 2022-06-11 by Asterchades

Triangulix, 2022-06-11:
The notation for calculated values starting with "=" exists at least since Cura 3.5. I could find it in default profiles of that version. From this version on these entries have been carried over through all 4.x versions to 5.0.