Ultimaker/Cura #12492: Skipping layers

Beeh1, 2022-06-10:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Testing out dual extrusion, I used a model that would demonstrate dual extrusion.[https://www.thingiverse.com/thing:2623645](url)
[Models used](https://github.com/Ultimaker/Cura/files/8881735/New.WinRAR.ZIP.archive.zip)

### Actual results

The model is mostly fine, except the middle portion of the inside part has layers being skippe
[image]
d

### Expected results

Uniform model, no layers skipped

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[log and project.zip](https://github.com/Ultimaker/Cura/files/8881759/log.and.project.zip)

GregValiant, 2022-06-11:
@Beeh1 That's an interesting model.
Go to the Mesh Fixes section and try turning off "Alternate Mesh Removal" then slice and check the preview.  The "Impossible Twist Inside" model looks like it will print separately i.e. as a separate un-attached part.  You may not need "Remove Mesh Intersections" at all.
So far this looks like "expected behavior".  Let me know if turning off the setting works.

Beeh1, 2022-06-11:
Turning off "Alternate Mesh Removal" fixed it. Taking a look at the impossible_twist_inside_2color.stl, it looks like it's actually in two parts, hence why that setting would show up. Thanks for the help!

GregValiant, 2022-06-11:
The loose center part should slide up and down after printing.  It's a desktop toy for busy executives?
I'll close this as it's fixed.

closed 2022-06-11 by GregValiant