Ultimaker/Cura #12496: The warning report with changed profile settings is incomplete

upatamby, 2022-06-10:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ultimaker 3E

### Reproduction steps

I choose a profile and then change some settings.
To view the changes I select a profile (no matter which) to view the report of changes.

Report changes:
[image]

### Actual results

The report doens't contain all changed profiles or even contains wrong settings (that were changed but ar not changed anymore).
To zee the write changes I have to close Cura and open it again (but then of course I also have to place and arrange the models again so that is obviously not what I want to do). 
Another problem is that there is no extruder number givenwith any of the profile settings, so you don't know to which extruder a specific setting belongs.

### Expected results

The Changes report should allways show the correct changes and if a setting beloings to a specific extruder that extruder must be mentioned (with consisting numbering - I prefer 1 and 2 and not 0 and 1, because that are the numbers that are used on the printer itself.

Let me explain why this is important to me:

When documenting my prints, I would like to save the Cura settings used and in particular the changes made to the selected profile. I can of course save the whole project, but that is not easy to read and it contains much more information than I need. I therefore use the following trick: I select a profile and wait for the changes report screen to appear.

In my documentation I copy this screen and also this screen-part:
[image]
so I know material, extruders and profile I used.

Of course, a separate function that copies all this data directly to the clipboard as readable text would be even more convenient, but in the absence of such a function, I do it as described.

And if it's not too much to ask, can't a button or menu item be added somewhere: "Copy changed settings to clipboard" with which the data from the above screen shots can be copied as text to the clipboard? 
I would find that extremely useful.

Thanks.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

GregValiant, 2022-06-11:
One of the big changes in 5.0 was the move from Qt5 dialog controls to Qt6 dialog controls.  I think they will keep evolving as the Cura Team get more experience with the new controls.  If Cura isn't loading the current variable settings into the box then that is probably something different.
Being able to copy the dialog information to the clipboard is something that I can also see as being useful.  I would view that as a "New Feature".
The "Per Extruder" issue would seem to be either one or the other.
This may be something the team will want to discuss so I'll mark it for them.

nallath, 2022-06-13:
I've added it to the list of things to discuss. Thanks!

nallath, 2022-06-13:
We've added this ticket to our backlog. For future reference: CURA-9270

GregValiant, 2022-06-13:
Jaime,
I came across an item on Github that I would like to change the title on.
It has changed from a bug to a feature request and the title could be more
descriptive of what the OP wants.  I was unable to get in and change it.
User error I think.  It's #12397.
It's something that Ghostkeeper said I would have access to.  If I don't
have access that's fine else let me know how.

Thanks,
Greg