Ultimaker/Cura #12497: cura 4.13.1 : spectacular deluge of error dialogs and crash

fenugrec, 2022-06-11:
### Application Version

4.13.1

### Platform

linux

### Printer

anycubic i3 mega s/pro

### Reproduction steps

- Fiddle around with a part, slice, and print via USB (this was quite bold, in retrospect)

### Actual results

About 15 min into the print, Cura starts spamming about 10 dialogs per second with roughly the same message, 

```
2022-06-10 21:31:52,365 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2022-06-10 21:31:52,365 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: TypeError: unable to convert a Python 'float' object to a C++ 'int' instance
```

Sorry, no screenshot, and nothing very clear in cura.log . I was in full abort mode to save my desktop session before Cura crashed X or my WM. I probably had over 100 dialogs open before I was able to regain control.

### Expected results

Not spam dialogs faster than a human can close them ! Bonus : ideally, not crash at all, but it's not a perfect world.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

The log file is probably not very helpful, but here it is anyway.

[cura.log](https://github.com/Ultimaker/Cura/files/8882936/cura.log)

GregValiant, 2022-06-11:
Thanks for the report @fenugrec .  USB printing isn't "maintained" as no Ultimaker printers use it, but I really haven't had a problem printing from Windows to my Ender 3 Pro over the USB in either 4.13 or 5.0.  I know other people have though.
The lines in the log:
2022-06-10 21:31:52,352 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2022-06-10 21:31:52,352 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: TypeError: unable to convert a Python 'float' object to a C++ 'int' instance
might indicate something different than a USB Printing problem.  I'll mark this as something for the Cura team to look at.

fieldOfView, 2022-06-11:
Are you using the AppImage, or another distribution of Cura?

This is an incompatibility of Cura 4.x with Python 3.10 (and newer). This bug has been fixed in Cura 5.0. It is also likely not present in the AppImage of Cura 4.13 (which includes Python 3.8)

fenugrec, 2022-06-12:
> Are you using the AppImage, or another distribution of Cura?

Sorry, forgot to specify; it is my distro's build (compiled thus https://github.com/archlinux/svntogit-community/blob/b9775b94857705fd44062ec3f70686558d9acd2e/trunk/PKGBUILD#L41 )

> This is an incompatibility of Cura 4.x with Python 3.10 (and newer)

Ah, thanks. Then I'll just wait for my distro to catch up with your official releases.

fieldOfView, 2022-06-12:
> Then I'll just wait for my distro to catch up with your official releases.

Or you could use the AppImage, which was not affected by this issue in the first place.

I'm closing this issue because the incompatibility has already been fixed in Cura 5.0.

closed 2022-06-12 by fieldOfView