Ultimaker/Cura #12498: Spiralize Outer Contour IGNORES all retraction

brianmi40, 2022-06-11:
### Application Version

5

### Platform

Windows 10

### Printer

Aquila X1

### Reproduction steps

Enable Spiralize Outer Contour to print a vase.

### Actual results

Printing anything with Spiralize Outer Contour always dribbles filament in the bottom layers when it moves to a new starting point.  Cura should obey the Retraction settings when printing the bottom layers, then retraction will be unnecesarry once the single layer spirals start.

### Expected results

Retraction should function in the base layers before switching to only printing the perimeter.  any vase can be used.  I was printing https://www.thingiverse.com/thing:570288, one of the SOLID FILLED versions since the others somehow drag strings across as it is spiralizing (may or not be a Cura bug, could be a design issue for bonus points check this out also).

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

You don't need a file, this bug happens on EVERY spiralization that has a bottom layer(s).  also your stupid bug report form here asks for a PROJECT FILE BUT DOES NOT RECOGNIZE THAT FILE TYPE, so you DON'T GET ONE
[image]
.

GregValiant, 2022-06-11:
Yes, the file has to be zipped to be posted.  It's a GitHub thing with a 25mb limit for the zipped file.
_"...one of the SOLID FILLED versions since the others somehow drag strings across as it is spiralizing (may or not be a Cura bug, could be a design issue for bonus points check this out also)"_
In order for spiralize to work the model must be solid.  An empty glass won't spiralize (it has inside walls facing air as well as outside walls facing air) but a glass that is solid to the brim will spiralize (only the outside wall faces air).

In regards to retractions, there might not be many but there should be a few.  5.0 makes all bottom layers "concentric" and so doesn't require many retractions.  There might be a retraction when moving from a skirt to the real print.  When moving from a brim to the print there might not be one.
When slicing a solid cylinder with 4 bottom layers I get this for the first 4 layers:
G1 F2100 E2092.23089                --retraction
;MESH:NONMESH
G0 F600 X65.814 Y114.944 Z0.8  --travel
G0 F9000 X114.825 Y115.089      --travel
;TIME_ELAPSED:849.397009        --travel
;LAYER:3
;TYPE:WALL-INNER
;MESH:cylinder.3mf
G1 F2100 E2097.73089                --un-retract
So it retracts only once per layer as everything else was a single continual movement.  There was no need (in that model) to end an extrusion and travel to a new point.
If your Z seam setting is on the opposite side from the Travel / Layer Start setting then there should be a retraction provided that your "Retract Minimum Travel" and "Max Comb Distance with no Retract" are set to allow a retraction given the distance traveled.
In my model what I see is "expected behavior".
Without your project file I can't tell what is going on with your vase.

jrutgeer, 2022-06-24:
> In order for spiralize to work the model must be solid. An empty glass won't spiralize (it has inside walls facing air as well as outside walls facing air) but a glass that is solid to the brim will spiralize (only the outside wall faces air).

Actually this seems to have changed from Cura 4.12.1 to Cura 5.0.0.
In Cura 4 it sliced ok. In Cura 5 it gives odd results.

E.g. consider this thin-walled tube (it's a simple open cylinder with wall thickness 0,8mm): [Tube.zip](https://github.com/Ultimaker/Cura/files/8976136/Tube.zip).

Cura 4:

[image]

Cura 5:

[image]

GregValiant, 2022-06-24:
It's the variable line width feature of 5.0.  In 4.x no adjustment could be made so your walls were a single pass and spiralize worked even with the Surface Mode set to Normal.  5.0 is capable of making adjustments to the line width and so it's able to correctly slice the model and as a result, spiralize doesn't work in normal mode on this model because the model has two walls facing air.  That is accounted for in 5.0 whereas it was not in 4.13 and earlier.  There are other settings affected by the re-write of the engine but things having to do with walls will involve more of a learning curve.

These are with 5.0
This is with Surface Mode set to Surface and the layer start adjusted to match the z seam position.  There are two paths to the shell.  1 for the outer wall facing outboard and the second path is for the outer wall facing in towards the center.  The blue lines are at the z seam.  Spiralize is turned on and as you can see it did not work as it did in 4.x versions.
[image]

This is with Remove All Holes enabled and Surface Mode set to Normal.  Now the part is solid thru-and-thru and spiralize works fine.  One change that was made to spiralize and is drawing a lot of flack was making the bottom layers all concentric.  Not many people are fond of that change (and I'm one of them).
[image]

I think it is now a more accurate representation of the model.  If the wall of this model were 2mm thick how should it slice?  Should spiralize work in that case?  If would if you Removed all Holes.
Learning how to move around in Cura and make adjustments is going to take a bit of time.  I'm certainly not there yet...but I'm getting better at seeing how things work in 5.0 and how to make adjustments from the way I was doing things to what is required in 5.0.  I much prefer 5.0 to the Arachne betas.  I think when problems with the dialogs go away and the "unable to slice" problems get resolved AND people get used to it - that it will result in better prints than the previous versions did and designing the parts for FDM won't be as fussy because the walls will tend to work out in Cura as opposed to contantly fooling around with the wall widths in CAD hoping to get a good strong model.

closed 2022-06-25 by github-actions[bot]