Ultimaker/Cura #12499: Preview Shows Incorrect Line Width

rachael7, 2022-06-11:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

1. Slice a model. Note the line width colors and scale on the preview screen.
2. Change material flow rate from 100% to any other number. 50% will make it obvious.
3. Check the line width colors and scale on the preview screen again. Note that they are different from before.

### Actual results

When the flow rate is reduced, the preview screen changes. The preview shows all the lines narrower by the same percentage as the flow rate reduction, even though there was no change to the line width setting. 

### Expected results

A change in flow rate alone should not change the displayed line width on the preview screen. Slicing twice, while changing only the flow rate should produce identical preview screens.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[Flow Rate Test Cubes.zip](https://github.com/Ultimaker/Cura/files/8883225/Flow.Rate.Test.Cubes.zip)

With the new variable line width, it becomes necessary to check the line widths Cura comes up with in different parts of the model. It is especially important as I am learning the impact of the new settings related to this. However, it seems that the preview screen in Cura is not showing the line width it derived, but rather the extrusion width you would get based on the geometric calculation of the flow rate. While this may or may not have any relation to the real life extrusion width, it is not at all the number I need to see to know if I have done well with my settings. I have some materials that print with flow rates as low as 50%, such as light weight (foaming) PLA. If I set a 0.4mm line width and preview, the widest line I see is 0.20mm, which is clearly incorrect. The attached project file has three test cubes set to print sequentially, each with a different flow rate, all other settings the same. I expected to see three identical cubes on the preview screen when using the line width color scheme, but instead, each cube shows in a different color. I also attached a screenshot showing that result.

GregValiant, 2022-06-11:
Thanks for the report.  The foaming PLA is a new material and may require some adjustments within Cura itself to print correctly.
I opened your project and took the liberty to change the 80% flow cube to 25%.  Then I sliced your project.  This image is the Gcode file read into Auto-Cad.  Measuring the line width spacing they all are .5 distance between them.
[image]

This image is from Cura in "compatibility mode".
For this slice I increased flow to 150% on the left model and reduced flow on the right model to 25%.  I think it's clear that there really is no difference in the index distance between the lines (Line Width) but rather each line is a lot thinner (Flow).  In the black blob you can see there is a lot of overlap between the inner and outer walls.  (My guess would be 50% overlap.)
Since my video system uses a CRT and vacuum tubes I can't check the OpenGL 4.1+ view that you have.

If you could post a gcode file I can take a look at that and see if the Line Width is actually being affected or if it is a screen rendering / video system issue.

rachael7, 2022-06-11:
Thanks for the reply and for checking it on your system. No need to send gcode - I can confirm that the actual line spacing is correct, at least in this simple case. This is strictly a matter of graphical representation on the preview screen. What we need to see on that screen is the line spacing, not how thick Cura thinks the extrusion might come out. The foaming PLA is only one example of a material where the actual line width is nothing like what you would calculate volumetrically, but the visualization problem is there anytime  you change the flow rates. What I'm looking for on the line width color scheme is the line spacing, to see if there are any problematic line widths anywhere in the print, but what Cura is showing me is something different entirely. I'm not sure what use the current scheme would be, but I really need the scheme where I can see the line spacing, without any graphical adjustments for flow rate. I think that's what the scheme was meant to show, no?

GregValiant, 2022-06-11:
In Cura, go to Preferences / Configure Cura and on the General tab under Viewport Behavior click on "Force Layer Compatibility Mode".  It will require that you close Cura and restart it.  You can save your work as a project file and then open that after the restart.  The old style 2d line view has some advantages over the 3d view and in this case might give you a better sense of what's going on with the line width.

fieldOfView, 2022-06-11:
> When the flow rate is reduced, the preview screen changes. 

This is by design. The preview shows what you are instructing your printer to lay down. Decreasing the flow means laying down less material, and this is what Cura shows. Cura does not know that the material expands after it has been extruded, and that this is what you intend.

I am closing this issue, because the behavior is "as designed". That does not mean that you can't continue discussing this here.

closed 2022-06-11 by fieldOfView

rachael7, 2022-06-11:
> This is by design. The preview shows what you are instructing your printer to lay down. Decreasing the flow means laying down less material, and this is what Cura shows. Cura does not know that the [...]

Hi @fieldOfView, 

Thank you very much for your reply and explanation. You hit on exactly my point though - Cura doesn't know what the filament will do after it comes out of the nozzle, the foaming material being but an extreme example of that effect. And since Cura doesn't know how the material will behave, it is useless to display what it thinks the extrusion width might be; it's a meaningless number that may or may not correspond in any way to the actual behavior of the machine and material. The line width Cura has determined, or if  you prefer the "line spacing", is a critical number that does correspond to the actual behavior of the machine and could need to be adjusted.

When I check the line widths in the preview, I'm not looking for the effective extrusion width of an idealized material in an idealized machine, I'm looking to see if any of the line widths are too low or too high to extrude well with the nozzle I'm using. So when I lower my flow rate to correct for material/machine behavior and Cura reports that as a narrower line width, I lose the information I need. If I see a line width of 0.2mm in the preview, I know that won't extrude well with a 0.4mm nozzle, so I want to make an adjustment. But if the real line width (or the line spacing, if you prefer) is 0.4 and the flow rate is 50% because of the material I'm using, that is actually fine and I don't need to make any adjustment to the line width. That is what I thought that preview color scheme was for, and certainly how I would like to use it.

I understand that the current behavior is as designed though, so I guess this has to be a feature request, rather than a bug report. Are you able to change the post to reflect that?  Thanks.

fieldOfView, 2022-06-13:
To start, you would need to describe the solution you would want (realizing that what you need for this particular usecase may not work for other usecases). Something like an "ignore flow" checkbox in the layer view?

A workaround is to set the flow back to 100%, slice and check the linewidths. Then if you are satisfied slice again with the original lower flow.