Ultimaker/Cura #12500: Immidietly crashing when application opens to a second monitor.

DaBuKeZ, 2022-06-11:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 3 v2

### Reproduction steps

1. Cura was crashing everytime i tried to open it 
2. cura was opening to a second monitor 

**I have two monitor setup where main monitor is in landscape mode and second one is in portait mode**

1. tried to unplug second monitor 
2. Cura opened normally
3. Pluged secod monitor back in 
4. Cura crashed when i tried to open it. 
5. switched second monitor to landscape mode had no effect to the problem. 
6. opening from different file types or just the program had no effect.

### Actual results

Cura crashed immediately and i cant use it until i unplug my second monitor. 

### Expected results

cura should open normally when i have two monitors pluged in. Problem started when i updated to cura 5.0.0. Cura 5.0.0 worked once afted the update. There hasn't been any changes to other drivers, hardware or software. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8883583/cura.log)

specs: 

AMD Ryzen 3700x 
16Gb Kingston hyperX DDR4 ram 
ASUS STRIX RTX 3080
C: is on NVME M.2 SSD

GregValiant, 2022-06-11:
This appears to be a duplicate of #12197 and so I'll mark it.
The "dual monitor" thing keeps coming up.  The developers all use dual monitors and none of them had an issue.  That means it's likely related to specific machines and hardware/software configurations.  I'll leave this open if you care to respond (or think your problem is different enough that this should be considered a different issue).  My thought on this is that it isn't an easy fix.  (I have noticed that the words "Asus" and "Ryzen" seem to come up a lot.)

fieldOfView, 2022-06-11:
This duplicate of #12197 can be closed.

closed 2022-06-11 by fieldOfView