Ultimaker/Cura #12501: Klipper Flavor

DvoraNoob, 2022-06-11:
### Is your feature request related to a problem?

My real problem is the estimated time, as it is very wrong, but there is another problem with using Marlin Flavor for Klipper, for example Klipper doesn't use jerk, and the acceleration is usually higher than in Marlin.

### Describe the solution you'd like

I think if you create a Klipper Flavor with the option to configure the settings as in the Klipper config file, and if the Klipper Flavor is selected by the user it would remove the Jerk settings and add the Square Corner Velocity instead, I think that would solve the problem. estimated time.

If possible add the shortcut to edit de Pressure Advance and Input Shaper

### Describe alternatives you've considered

Currently I'm using the Printer Settings plugin to modify the maximum acceleration and maximum speed, but the plugin only has jerk so the estimated time is better, but I don't like using Marlin, another thing I'm using is the Klipper Estimator, but it only updates the time when I send GCode to Klipper.

### Affected users and/or printers

Everyone who uses Klipper

fieldOfView, 2022-06-11:
I would guess that 98% of development of Cura is done by Ultimaker developers, who are paid by Ultimaker. Ultimaker does make a printer that uses/supports Klipper, so it is unlikely that Ultimaker developers will put time into this. Not just because they are not paid to do so, but also because they don't have any printers to test it with. 

You are likely going to have to rely on someone implementing this and creating a PR, or someone like me to make a plugin for Cura to make this happen.

Sonicboom247, 2023-02-21:
@fieldOfView love your plugins by the way!,so helpful! That being said i would also love to see this or something similar implemented. 3 of my printers are Marlin and one is Klipper and although I use Cura for all, when it comes to Klipper there are so many hoops to jump through..

closed 2024-03-06 by github-actions[bot]