Ultimaker/Cura #12502: Auto Zoom Extents

RalfZgeib, 2022-06-11:
### Is your feature request related to a problem?

Always frustrated when I cant zoom to the exact extents of the object or generated gcode.

### Describe the solution you'd like

A button that would zoom and center the bounding extents of the object or toolpath

### Describe alternatives you've considered

No alternative found other than scrolling back and forth

### Affected users and/or printers

all users. no printers.

nallath, 2022-06-13:
I've added the feature request to the list to be discussed by the rest of the team. I will keep you updated!

nallath, 2022-06-21:
Due to the large number of issues already on our backlog, we've decided to defer this issue. This doesn't mean that we don't believe that this isn't a good issue to have, just that we have to prioritize what we do (and don't do).

If anyone would make a PR for this, we would of course welcome the contribution.

RalfZgeib, 2023-06-29:
Hello,

Thank you for the follow-up!

This issue is still an important feature that helps tremendously in productivity when using Cura.

It would be great if it can still be address!

Thanks a ton!

Best,
Ralf

Ralf Zgeib
+1(201)993-9709