Ultimaker/Cura #12503: Spiralize outer controu

thetechy2, 2022-06-12:
### Application Version

5.0.0

### Platform

windows 10

### Printer

custom i3

### Reproduction steps

trying to use vase mode with a partial surface model, exported from inventor 2021. top portion is only 1 surface, theads on the bottom is solid and when printing with spiralize outer contour and smooth spiral contour both enabled and it doesnt only print the outer contour at all it adds extra walls everywhere

### Actual results

same printer profile, same print settings in 5.0;
[image]

### Expected results

spiralize outer contour should only print the outer contour correct? this is in 4.11.10
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Project file-[foldy lampshade.zip](https://github.com/Ultimaker/Cura/files/8885188/foldy.lampshade.zip)
Log file-[cura.log](https://github.com/Ultimaker/Cura/files/8885191/cura.log)

GregValiant, 2022-06-12:
[image]

Thanks for the report.
You should have gotten that message when you loaded the file.  The purple areas indicate errors in the model.  When you sliced it, Cura got confused, and the model didn't slice the way you thought it should.
To use spiralize:  the model must be a solid all the way through.  Yours has a big ole hole down through the center.  You can turn on "Remove all Holes" in Cura and then it will spiralize, but there are still issues because of the errors in the model.
In the image below you can see disconnected artifacts hanging in the air near the top of the model.  You need to go back to Inventor and make adjustments.
I'm going to close this as at this point it's an obvious model problem.  If you feel it should remain open you can add a comment and I'll revisit this.
[image]

closed 2022-06-12 by GregValiant

thetechy2, 2022-06-12:
what version of cura are you using and how am i supposed to fix this? you say cura got confused but how tf was i supposed to know that you cant use a tube to spiralize the outer contour? ive literally been doing this for years. what is the differenct between a spiralized outer contour and a single wall model with no retractions anyways!? I have the settings guide extension and none of that is in it. There really needs to be a proper write up if you expect people to actually be able to use it properly. and how are those artifacts as you call them disconnected? in inventor thore are all fillets.