Ultimaker/Cura #12504: Spiralize outer contour problems with accuracy

thetechy2, 2022-06-12:
### Application Version

4.11.10

### Platform

WIN10

### Printer

custom i3

### Reproduction steps

put in model and set to vase mode print profile spiralize outer contour, smooth outer. the slicer just add random spots in the wall that arent present anywhere on my model.

### Actual results

These errors are barely visible from here
[image]
starting to see
[image]
there! it just moves a part of the wall there for no reason
[image]

### Expected results

this is what the model is supposed to look like
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Project-[foldy_lampshade.zip](https://github.com/Ultimaker/Cura/files/8885210/foldy_lampshade.zip)
Log-[cura.log](https://github.com/Ultimaker/Cura/files/8885211/cura.log)

GregValiant, 2022-06-12:
This is really a duplicate of #12503.  The model has errors, and is not suitable to use spiralize as it isn't solid all the way through and consists of two meshes.
The preview you are getting is not of a spiralized model but a conventionally sliced model with a single wall.  Your 1.0 line width may be hiding how the slicing really turned out.
I'll leave this one open if you would care to add further comments but I don't see a bug here.

nallath, 2022-06-13:
Furthermore, this is also done in cura 4.11, which will no longer get any updates.

closed 2022-06-13 by nallath