Ultimaker/Cura #12505: Cura 5 Stable crashes on Mojave 10.14.6

T-SquaredProductions, 2022-06-12:
### Application Version

5.0.0

### Platform

Mac OS 10.14.6

### Printer

None

### Reproduction steps

Downloaded Cura 5.0.

### Actual results

Cura does not open, and simply quits without a crash report. (Also does not leave a crash log file in the Application Support file)

### Expected results

Cura should open without a crash.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

(Unable to provide due to crash not leaving a log report)

T-SquaredProductions, 2022-06-12:
I apologize that there isn't any information on this. It really is that bad of a crash that it doesn't even start up. (Simply clicking the icon makes it bounce once, but then it completely stops.)

fieldOfView, 2022-06-12:
From the releasenotes of Cura 5.0 (https://github.com/Ultimaker/Cura/releases/tag/5.0.0):

> From version 5.0 onwards - Ultimaker Cura is only supported on operating systems actively maintained by their software manufacturer or community. This means Windows 7 and MacOS 10.14 will no longer be supported.

It would be better if the application outright stated this when it is launched on an unsupported system. I'll leave it up to the Ultimaker devs to decide if that is something they want to put time towards (if only to reduce support calls).

nallath, 2022-06-13:
> It would be better if the application outright stated this when it is launched on an unsupported system. I'll leave it up to the Ultimaker devs to decide if that is something they want to put time towards (if only to reduce support calls).

Yeah, agreed. Not sure if we can get that warning in all cases, but at least some would be better than none.

ctzsnooze, 2022-06-14:
Yes can confirm it just fails to open in Mojave / 10.14.  That's sad, I really like this OS.
Nothing happens at all, exits with no warning or explanation.
It would be really good to have it work on 10.14.  I have a lot of plugins etc that will not work when I go to newer firmware.

CaptainMalu, 2022-06-29:
Put a note near the download button on the ultimaker website. (Or add it to the system requirements page, but I'm not sure someone will follow this link).
That's also not foolproof but an easy way.

Maoman1, 2022-09-16:
Extremely disappointed in the lack of support for microsoft's best OS to date.

Darrenmeehan, 2022-09-18:
I also cannot open Cura Slicer. I am running Mac OS 10.15.7 which is the latest version of MacOS available for by Macbook Pro 2012.

philipphirte, 2022-11-03:
Same problem here