Ultimaker/Cura #12506: Slow to slice.

chibastos, 2022-06-12:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Creality Ender-3 MAX

### Reproduction steps

Slow to slice, takes longer than 30 minutes

### Actual results

Slow to slice, takes longer than 30 minutes

### Expected results

Slow to slice, takes longer than 30 minutes

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Slow to slice, takes longer than 30 minutes

fieldOfView, 2022-06-12:
Without a project file there is nothing we can do about this issue.

chibastos, 2022-06-12:
https://www.myminifactory.com/object/3d-print-deadpool-bust-rem
[image]
astered-supportless-edition-81435#google_vignette

chibastos, 2022-06-12:
I attached an image and the file path.

https://www.myminifactory.com/object/3d-print-deadpool-bust-remastered-supportless-edition-81435#google_vignette
[https://cdn.myminifactory.com/assets/object-assets/5c145d5a14bbc/images/720X720-deadpool-bust-v2-01.jpg]<https://www.myminifactory.com/object/3d-print-deadpool-bust-remastered-supportless-edition-81435#google_vignette>
3D Printable Deadpool Bust (Remastered Supportless Edition) by Eastman - MyMiniFactory<https://www.myminifactory.com/object/3d-print-deadpool-bust-remastered-supportless-edition-81435#google_vignette>
| Download free and paid 3D printable STL files
www.myminifactory.com

fieldOfView, 2022-06-12:
None of the links you posted work. From the image, it looks like a fairly detailed model, and there are certain settings that are computationally intensive to slice. You could try slicing with regular supports instead of tree supports.

To get a definitive answer, we really need a project file, because that includes all your settings. A project file can be created with File -> Save project...

chibastos, 2022-06-12:
[https://res.cdn.office.net/assets/mail/file-icon/png/generic_16x16.png]deadpool-bust-v2-supportless.stl<https://1drv.ms/u/s!AjAsASc_dmRZmHWFf3a-EHLghhhF>[https://res.cdn.office.net/assets/mail/file-icon/png/generic_16x16.png]deadpool-bust-v2-supportless.3mf<https://1drv.ms/u/s!AjAsASc_dmRZmHS4MV085Noh5Gt5>

in cura 4.13.1 I'm getting it faster

closed 2022-06-27 by github-actions[bot]