Ultimaker/Cura #12508: Under extrusion at Z seam on Creality printer

wren945, 2022-06-12:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality CR-6 SE

### Reproduction steps

 - Select printer profile for Creality CR-6 SE.
 - Use regular PLA material and select Generic PLA material preset.
 - Use Standard Quality - 0.2mm profile, and don't change any settings.
 - Print THE calibration cube (https://www.thingiverse.com/thing:1278865)
 - See the attached picture for severe under extrusion at layer change location ( Z seam).

[image]

Also attached is the project file.
[Project.zip](https://github.com/Ultimaker/Cura/files/8885931/Project.zip)

### Actual results

[image]

### Expected results

The previous version has no such problem.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project file:
[Project.zip](https://github.com/Ultimaker/Cura/files/8885932/Project.zip)

GregValiant, 2022-06-12:
Hello @wren945 Thanks for the report.
The settings aren't optimized for a particular printer.  One of the settings that my Ender 3 Pro truly hates is "Retract at layer change" and another is "Retract before outer wall".  When there is a retraction - travel - extrusion start the nozzle can ooze during the travel and it takes longer build up the pressure in the nozzle than was anticipated.  The nozzle always starts to travel immediately upon completion of the prime move.  If the pressure isn't up then extrusion doesn't start when it should.
So I suggest you turn off "Retract Before Outer Wall" and maybe print a retraction tower to help dial in your retract and prime speeds.
After opening your project file I changed the print temperature to 205 and the bed temperature to 50 then sliced and printed it.  I left "retract at layer change" turned on.  You can see that there is some sort of artifact about half way up but it is not at the prime position, but shortly after.  (Under the microscope it's black so maybe something cleared out of the nozzle.)
For most prints I consider 3 walls to be a minimum and 4 tops/bottoms to be a minimum.  When you get below those numbers then infill can leave marks on the outer walls and on top surfaces.
So I suggest a couple of changes to your settings and try again.  I'll leave this open but I don't see a bug here.
[image]

altbanky, 2022-06-13:
I am having the same issue with Z seam on the Ender 3 S1 Pro. 

Just tried @GregValiant suggestions and toggled both 1) Retract at layer change and 2) Retract before outer wall with no succeess.

I switched to Cura 4.13 and it worked with the same settings.

skl111, 2022-06-13:
This may be related to this issue.
https://github.com/Ultimaker/Cura/issues/12026

Asterchades, 2022-06-13:
When comparing to earlier Cura versions, make sure to compare the _exact_ same settings. Cura 5 has changed the default Wall Ordering - earlier versions are Inside To Outside, while 5.0 is now Outside to Inside. Doing the walls in the reverse order, in tandem with Optimize Wall Printing Order, will effectively use the Inner Wall(s) as a "prime line" for the Outer Wall. In other words the issue is probably still there - just hidden inside the model instead of exposed.

richfelker, 2022-06-16:
I commented on the linked issue with what I believe is a plausible explanation and solution: https://github.com/Ultimaker/Cura/issues/12026#issuecomment-1157801681

wren945, 2022-10-11:
> explanation and solution: #12026 (comment)

Thanks a lot. I'm going to give it a try during the weekend.