Ultimaker/Cura #12509: Bottom pattern same for Lines and Zig Zag

floridajohn, 2022-06-12:
### Application Version

5.0.0

### Platform

Windows 10 Pro - Xeon E5 2690 - HP Z620 

### Printer

Creality Ender 3

### Reproduction steps

Large print (~80mm x 230mm) that I actually had to rotate it on the plate (~45 degrees to get it to fit - had to turn off snap to get it exactly in the orientation that would print.

Lines was producing a rough looking bottom surface.  Switched to Zig Zag but the physical output was still Lines.  Preview for both is the same.  Concentric preview is as expected as is the print.

I've only been doing 3D printing for 5 months so please have some patience with me - and this is my first CURA ticket.

[C6 coffee V2.zip](https://github.com/Ultimaker/Cura/files/8886007/C6.coffee.V2.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/8886009/cura.log)

### Actual results

Lines not Zig Zag for bottom layers

### Expected results

Zig Zag not lines for bottom layers

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Low end video card - NVIDIA GeForce 9800 GTX/GTX+

Asterchades, 2022-06-13:
Are you able to provide screenshots of the (mis)behaviour? I'm unable to reproduce what you're describing using the project file you've given.

Zig Zag:
[image]

Lines:
[image]

Both are similar as expected, but you can clearly see the links between the lines around the circle and along the edges when the Zig Zag pattern is in use.

floridajohn, 2022-06-13:
I actually physically printed them part way.  Went out to get them from the
trash but they got an early start and they are gone.  Attached are 3 pics.

Asterchades, 2022-06-13:
Your images haven't attached. You might need to upload them directly via GitHub rather than replying to the email notification with attachments.

I did double check just in case the preview is showing the links and the GCode isn't actually being generated and Cura, PursaSlicer , and IdeaMaker's GCode viewers are showing the joining lines present. I can't actually run a test print as my printer is physically too small but something very odd would need to be happening at the printer level for the GCode to be present but not executed.

floridajohn, 2022-06-13:
It's my first GitHub.  I just attached them to a reply email.  Here you go:

[image]
[image]
[image]
.

closed 2022-06-14 by floridajohn

floridajohn, 2022-10-11:
> You misunderstand me. I don't mean zoom in the picture - I mean zoom in on
> the model preview! You can use the mouse scroll wheel in the Cura UI to
> zoom in and out on the preview screen, letting you see the actual extrusion
> lines rather than just a moiré pattern. That way resolution doesn't make
> any difference, nor does the presence or absence of a KVM switch or
> whatever video interface is in use.
>
> If you're wondering what I'm trying to get at with these requests, I'm
> trying to establish what you feel like Zig Zag should be doing versus what
> it's actually doing, in addition to whether it's actually doing what it's
> supposed to be doing. Lines and Zig Zag are *extremely* similar - so much
> so that if you just look at them from a distance, or if you look at them
> within the middle of a consistent surface, they will look identical to each
> other. It's *only* those little connecting links along the walls which
> separate the two, so you need to be able to see what it's doing against
> those walls in order to determine if it's working and if it's what you
> expect.

Ok, thanks for the clarification.  Let's let this go.