Ultimaker/Cura #12510: Cura 5 crashes on opening marketplace

coloso1967, 2022-06-12:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

opened cura clicked on marketplace to update apps

### Actual results

cura froze. waited several minutes but would not respond

### Expected results

marketplace opens and allows me to update app
[image]
s

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

does not seem to matter what machine i have cura on it freezes on all of them i have older cura that works fine

nallath, 2022-06-13:
Do you have an external monitor? Could you also share the log files?

Cpt-Marcelino, 2022-06-15:
I also have the problem.
I use 2 monitor

[cura.log](https://github.com/Ultimaker/Cura/files/8913049/cura.log)

coloso1967, 2022-06-18:
here is my cura log and yes i have 2 monitors and a sensor panel connected to my P
[cura.log](https://github.com/Ultimaker/Cura/files/8932835/cura.log)
C

mrbass21, 2022-06-26:
I also have 3 monitors (2560x1440) (Windows 10) and any time Cura or any window Cura launches is not on the primary display, Cura crashes.
 I had to turn off Restore window position on start because the application would just immediately crash, however marketplace or any other sub UI component that Cura launches and is not on the primary display crashes.

talentedmrripley, 2022-07-05:
I just downloaded the 5.1 beta and this is still occurring.  The Marketplace window opens but remains blank and shortly the title bar is updated with (Not Responding).  Attempting to close it causes Cura to exit entirely.

coloso1967, 2022-07-08:
i found the answer that worked for me on youtube https://www.youtube.com/watch?v=XW-E1P6Ms4s create with tech this guy walks you through the process of shutting down the service that is causing the issue and for me at least it worked.

jonfairbanks, 2022-07-09:
Thanks @coloso1967. 

TLDW: On Windows 11 (probably 10 too) stop the Nahimic service and set the startup type to manual.

nallath, 2022-07-15:
> It also should give good direction to the devs on how to reproduce. :)

It does, but it doesn't mean that it's something that we can actually solve. As we're not the only one that has this issue, it's very likely to be an issue with Nahimic

talentedmrripley, 2022-08-25:
An update, this is still an issue with the full release of 5.1.0.

talentedmrripley, 2022-11-16:
Another update, with 5.2.1 this is still occurring.

JimboFlowe, 2023-06-06:
Issue with 5.3.1 as well.